VMaxx/RepetierIntegration #1: Stuck when entering API Key

ebrithiljonas, 2018-01-31:
When I enter the API Key a completely empty windows pops up and nothing is happening. The other window where I entered the KEy just says "checking API key". I'm using Cura 3.1

eric-void, 2018-02-03:
I confirm the plugins don't work on Cura 3.1.

It gives a lot of errors. The first one (as said in previous post) is when entering API key.
This is the error:
`2018-02-02 22:51:39,872 - DEBUG - RepetierPlugin.DiscoverRepetierAction.testApiKey [265]: Trying to access Repetier instance at http://127.0.0.1:3344/ with the provided API key.
2018-02-02 22:51:39,873 - DEBUG - RepetierPlugin.DiscoverRepetierAction.testApiKey [266]: Using None as work_id
2018-02-02 22:51:39,877 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-02 22:51:39,878 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-02 22:51:39,878 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/DiscoverRepetierAction.py", line 269, in testApiKey
2018-02-02 22:51:39,878 - CRITICAL - cura.CrashHandler.__init__ [65]:     url = QUrl(base_url + "printer/api/" + work_id + "?a=getPrinterConfig&apikey=" + api_key)
2018-02-02 22:51:39,878 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: must be str, not NoneType
`

I tryed to insert manually "work_id" in the code, so the procedure goes on.
But it breaks later, when reading JSON from repetier.
It gives a lot of these errors:
`
2018-02-02 22:58:45,825 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-02 22:58:45,826 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-02 22:58:45,826 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 649, in _onRequestFinished
2018-02-02 22:58:45,826 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "numExtruder" in json_data[self._key]:
2018-02-02 22:58:45,826 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 'NAME'
2018-02-02 22:58:45,827 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-02 22:58:45,827 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-02 22:58:45,827 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 700, in _onRequestFinished
2018-02-02 22:58:45,827 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "job" in json_data[0]:
2018-02-02 22:58:45,827 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 0
`
... and a lot more ...

This error is also logged sometimes:
file:///usr/lib/cura/plugins/RepetierPlugin/DiscoverRepetierAction.qml:236: TypeError: Cannot read property 'baseURL' of undefined

VMaxx, 2018-02-04:
I'll have to try printing to myself.  I use a raspberry pi so there may be a difference there

eric-void, 2018-02-07:
With these settings the api_key input field is filled. But when i press the button "Connect" several empty windows opens, until Cura crashes.

Console log this:

2018-02-07 23:06:35,150 - DEBUG - RepetierPlugin.RepetierOutputDevice.connect [346]: Connection with instance Golconda with url http://127.0.0.1:3344/ started
2018-02-07 23:06:35,153 - DEBUG - RepetierPlugin.RepetierOutputDevice.connect [346]: Connection with instance Golconda with url http://127.0.0.1:3344/ started
2018-02-07 23:06:35,165 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-07 23:06:35,165 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-07 23:06:35,165 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 649, in _onRequestFinished
2018-02-07 23:06:35,166 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "numExtruder" in json_data[self._key]:
2018-02-07 23:06:35,166 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 'Golconda'
2018-02-07 23:06:35,167 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-07 23:06:35,167 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-07 23:06:35,167 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 700, in _onRequestFinished
2018-02-07 23:06:35,167 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "job" in json_data[0]:
2018-02-07 23:06:35,167 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 0
2018-02-07 23:06:35,168 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-07 23:06:35,168 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-07 23:06:35,168 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 649, in _onRequestFinished
2018-02-07 23:06:35,169 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "numExtruder" in json_data[self._key]:
2018-02-07 23:06:35,169 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 'Golconda'
2018-02-07 23:06:37,056 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-07 23:06:37,056 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-07 23:06:37,056 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 700, in _onRequestFinished
2018-02-07 23:06:37,057 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "job" in json_data[0]:
2018-02-07 23:06:37,057 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 0

... and other lines like these ...

VMaxx, 2018-02-08:
I posted an update that should resolve the key issue.  The only file to change was DiscoverRepetierAction.py  For eric-void, there may be more for me to fix for yours, as I only had one extruder the code may not be correct if you have more than one.  I'll have to create a dummy server with extra extruders and see if I get the same error.

VMaxx, 2018-02-08:
eric-void please try the new update if you have 3.1 it appears to work ok with two extruders in my testing.

eric-void, 2018-02-12:
The problem with the work_id now is fixed, but other errors persist :(

018-02-12 21:29:42,082 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-12 21:29:42,083 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-12 21:29:42,083 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 700, in _onRequestFinished
2018-02-12 21:29:42,083 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "job" in json_data[0]:
2018-02-12 21:29:42,085 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 0
2018-02-12 21:29:42,086 - INFO - UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device Golconda already added
2018-02-12 21:29:42,087 - INFO - UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device Golconda already added
2018-02-12 21:29:42,087 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-12 21:29:42,088 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-12 21:29:42,088 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 649, in _onRequestFinished
2018-02-12 21:29:42,089 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "numExtruder" in json_data[self._key]:
2018-02-12 21:29:42,089 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 'Golconda'
2018-02-12 21:29:42,090 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2018-02-12 21:29:42,090 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-02-12 21:29:42,090 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/RepetierPlugin/RepetierOutputDevice.py", line 649, in _onRequestFinished
2018-02-12 21:29:42,091 - CRITICAL - cura.CrashHandler.__init__ [65]:     if "numExtruder" in json_data[self._key]:
2018-02-12 21:29:42,091 - CRITICAL - cura.CrashHandler.__init__ [65]: KeyError: 'Golconda'
2018-02-12 21:29:44,037 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!

eric-void, 2018-02-12:
This is the json exported by the api call, maybe could be useful:

```
{"connection":{"lcdTimeMode":4,"resetScript":"","serial":{"baudrate":115200,"communicationTimeout":30.000000000000000,"device":"/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A92PXN3J-if00-port0","inputBufferSize":127,"pingPong":false}},"extruders":[{"acceleration":5000.0000000000000,"alias":"","changeFastDistance":20.000000000000000,"changeSlowDistance":20.000000000000000,"cooldownPerSecond":0.50000000000000000,"eJerk":20.000000000000000,"extrudeSpeed":2.0000000000000000,"filamentDiameter":1.7500000000000000,"heatupPerSecond":2.0000000000000000,"lastTemp":195,"maxSpeed":50.000000000000000,"maxTemp":260,"offsetX":0.0000000000000000,"offsetY":0.0000000000000000,"retractSpeed":30.000000000000000,"temperatures":[{"name":"ABS 245","temp":245},{"name":"PLA 210","temp":210},{"name":"PLA 195","temp":195}]},{"acceleration":5000.0000000000000,"alias":"","changeFastDistance":20.000000000000000,"changeSlowDistance":20.000000000000000,"cooldownPerSecond":0.50000000000000000,"eJerk":20.000000000000000,"extrudeSpeed":2.0000000000000000,"filamentDiameter":1.7500000000000000,"heatupPerSecond":2.0000000000000000,"lastTemp":180,"maxSpeed":50.000000000000000,"maxTemp":260,"offsetX":0.0000000000000000,"offsetY":0.0000000000000000,"retractSpeed":30.000000000000000,"temperatures":[{"name":"ABS 245","temp":245},{"name":"PLA 210","temp":210},{"name":"PLA 195","temp":195}]}],"general":{"active":true,"defaultVolumetric":false,"eepromType":"repetier","fan":true,"fan2":false,"firmwareName":"Repetier-Firmware","heatedBed":true,"logHistory":true,"name":"Tevo Tarantula","pauseHandling":0,"pauseSeconds":120,"printerVariant":"cartesian","sdcard":true,"slug":"Tevo_Tarantula","softwarePower":false,"tempUpdateEvery":1,"useModelFromSlug":"","useOwnModelRepository":true},"heatedBed":{"cooldownPerSecond":0.020000000000000000,"heatupPerSecond":0.20000000000000001,"installed":true,"lastTemp":0,"maxTemp":120,"temperatures":[{"name":"ABS","temp":115},{"name":"PLA","temp":50}]},"heatedChamber":{"installed":false,"maxTemp":100},"movement":{"G10Distance":3.0000000000000000,"G10LongDistance":13.000000000000000,"G10Speed":40.000000000000000,"G10ZLift":0.0000000000000000,"G11ExtraDistance":0.0000000000000000,"G11ExtraLongDistance":0.0000000000000000,"G11Speed":20.000000000000000,"allEndstops":true,"invertX":false,"invertY":false,"invertZ":false,"maxXYSpeed":250.00000000000000,"maxZSpeed":2.0000000000000000,"movebuffer":16,"timeMultiplier":1.0000000000000000,"xEndstop":true,"xHome":0.0000000000000000,"xMax":200.00000000000000,"xMin":0.0000000000000000,"xyJerk":8.0000000000000000,"xyPrintAcceleration":800.00000000000000,"xySpeed":100.00000000000000,"xyTravelAcceleration":3000.0000000000000,"yEndstop":true,"yHome":0.0000000000000000,"yMax":200.00000000000000,"yMin":0.0000000000000000,"zEndstop":true,"zHome":0.0000000000000000,"zJerk":0.30000000000000004,"zMax":200.00000000000000,"zMin":0.0000000000000000,"zPrintAcceleration":100.00000000000000,"zSpeed":2.0000000000000000,"zTravelAcceleration":100.00000000000000},"properties":{},"quickCommands":[],"shape":{"basicShape":{"color":"#dddddd","radius":100,"shape":"rectangle","x":0,"xMax":200.00000000000000,"xMin":0.0000000000000000,"y":0,"yMax":260.00000000000000,"yMin":0.0000000000000000},"gridColor":"#454545","gridSpacing":10.000000000000000,"marker":[]},"webcam":{"dynamicUrl":"","method":0,"orientation":0,"reloadInterval":3.0000000000000000,"staticUrl":"","timelapseBitrate":1000,"timelapseFramerate":30,"timelapseHeight":0.10000000000000001,"timelapseInterval":20.000000000000000,"timelapseLayer":1,"timelapseMethod":0,"timelapseSelected":0}}
```

VMaxx, 2018-02-12:
Send me what you get from stateList.  It gets the number of extruders from that.

eric-void, 2018-02-13:
Here it is:
```
{"Tevo_Tarantula_Dual":{"activeExtruder":0,"debugLevel":6,"extruder":[{"error":0,"output":0,"tempRead":15.000000000000000,"tempSet":0.0000000000000000},{"error":0,"output":0,"tempRead":14.699999809265137,"tempSet":0.0000000000000000}],"fan2On":false,"fan2Voltage":0,"fanOn":false,"fanVoltage":255,"firmware":"Repetier_0.92.8","firmwareURL":"https://github.com/repetier/Repetier-Firmware/","flowMultiply":100,"hasXHome":true,"hasYHome":false,"hasZHome":true,"heatedBed":{"error":0,"output":0,"tempRead":15.000000000000000,"tempSet":0.0000000000000000},"layer":198,"lights":0,"numExtruder":2,"powerOn":false,"rec":false,"sdcardMounted":true,"speedMultiply":100,"volumetric":false,"x":79.237998962402344,"y":200.00000000000000,"z":20.500000000000000}}
```

VMaxx, 2018-02-13:
The name of your printer is Tevo_Tarantula_Dual and not Golconda, I have the instance name and the internal name the same.  I think that's where things are different.  I'll look into what I can do with that.

eric-void, 2018-02-13:
Yes, "Golconda" is the name of the server profile on Cura, "Tevo_Tarantula_Dual" is the name of the machine on repetier.
If you think i can solve by renaming the Cura server to "Tevo_Tarantula_Dual" i can give it a try.

VMaxx, 2018-02-14:
I updated it to have some error handing to not dump cura.  That way you can adjust it.  The printer name in cura can be whatever, the main thing is when you set the name in the plugin, it has to match what is configured in Repetier Server.

VMaxx, 2018-02-16:
I will close this issue if you're happy it's resolved.

eric-void, 2018-02-18:
Ok, now it's working!
But i had to change the name of the repetier instance to "Tevo_Tarantula_Dual" (the name of the printer on repetier). Not doing this will result in "repetier_id = Golconda" written on [machine].global.cfg file, and the connection fails.

By renaming the repetier instance to the name of the printer it seems to work ok.

Thank you!
You can close the issue for me.

VMaxx, 2018-02-21:
Awesome thanks for replying.

closed 2018-02-21 by VMaxx