VMaxx/RepetierIntegration #2: Crash in Cura 3.2

B3DTech, 2018-02-06:
Hi - running Cura 3.2 on the Mac and the plugin crashes when entering the API key.  As soon as you paste the API key it crashes.

VMaxx, 2018-02-08:
I've not tested Cura 3.2 yet, I'm told there was a major rewrite of the plugins though so it may take some time for me to resolve.

VMaxx, 2018-02-08:
I created a new branch for 3.2 Cura.  Give it a try.

badvision, 2018-02-11:
It crashes when I try on 3.2 / Windows, even if I run the program as administrator (so that would nix it being a permissions issue) -- I tracked down the cura log file and found this:

`2018-02-11 10:45:59,715 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-11 10:45:59,716 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-11 10:45:59,717 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]:   File "C:\Program Files\Ultimaker Cura 3.2\plugins\RepetierPlugin\RepetierOutputDevice.py", line 655, in _onRequestFinished
2018-02-11 10:45:59,719 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]:     if "numExtruder" in json_data[self._key]:
2018-02-11 10:45:59,720 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]: KeyError: 'KriegerKloneMacher'
2018-02-11 10:45:59,732 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-11 10:45:59,734 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-11 10:45:59,738 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]:   File "C:\Program Files\Ultimaker Cura 3.2\plugins\RepetierPlugin\RepetierOutputDevice.py", line 655, in _onRequestFinished
2018-02-11 10:45:59,740 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]:     if "numExtruder" in json_data[self._key]:
2018-02-11 10:45:59,742 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]: KeyError: 'KriegerKloneMacher'
2018-02-11 10:45:59,743 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-11 10:45:59,745 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-11 10:45:59,746 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]:   File "C:\Program Files\Ultimaker Cura 3.2\plugins\RepetierPlugin\RepetierOutputDevice.py", line 706, in _onRequestFinished
2018-02-11 10:45:59,747 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]:     if "job" in json_data[0]:
2018-02-11 10:45:59,748 - CRITICAL - [(51192)-MainThread] cura.CrashHandler.__init__ [67]: KeyError: 0`

badvision, 2018-02-11:
Also seeing this, and I believe the issue is the url being formulated has the wrong thing in it for the printer in question (aka Custom FDM Printer): ```2018-02-11 10:45:50,224 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction.testApiKey [260]: Trying to access Repetier instance at http://192.168.1.66:3344/ with the provided API key.
2018-02-11 10:45:50,225 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction.testApiKey [261]: Using Custom FDM printer as work_id
2018-02-11 10:45:50,226 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction.testApiKey [262]: Using [SOME_API_KEY] as api_key
2018-02-11 10:45:51,560 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction._onRequestFinished [218]: API key accepted by Repetier.
2018-02-11 10:45:51,560 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction._onRequestFinished [223]: http://192.168.1.66:3344/printer/api/Custom FDM printer?a=getPrinterConfig&apikey=[SOME_API_KEY]
2018-02-11 10:45:51,560 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction._onRequestFinished [224]: {'error': 'Unknown printer'}
2018-02-11 10:45:53,654 - DEBUG - [(51192)-MainThread] AutoSave.AutoSave._onTimeout [71]: Autosaving preferences, instances and profiles
2018-02-11 10:45:57,584 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction.testApiKey [260]: Trying to access Repetier instance at http://192.168.1.66:3344/ with the provided API key.
2018-02-11 10:45:57,585 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction.testApiKey [261]: Using Custom FDM printer as work_id
2018-02-11 10:45:57,585 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction.testApiKey [262]: Using [SOME_API_KEY] as api_key
2018-02-11 10:45:57,614 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction._onRequestFinished [218]: API key accepted by Repetier.
2018-02-11 10:45:57,615 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction._onRequestFinished [223]: http://192.168.1.66:3344/printer/api/Custom FDM printer?a=getPrinterConfig&apikey=[SOME_API_KEY]
2018-02-11 10:45:57,615 - DEBUG - [(51192)-MainThread] RepetierPlugin.DiscoverRepetierAction._onRequestFinished [224]: {'error': 'Unknown printer'}```

badvision, 2018-02-11:
It seems like it is using the Cura "Printer Type" as the name of the printer queue on Repetier Server not the name of the printer or the name of the Repetier configuration (which that would be best I think.)   Second to that, you should probably wrap the affected lines with an exception handler so it doesn't kill the whole program, but instead just spit out errors to the log and say "Something didn't work, check logs" or whatever.

badvision, 2018-02-11:
Here's what I get from the printerConfig API from my copy of RepetierServer (0.86.2) which doesn't have "numExtruders" in the output as your program is expecting:

```{
"connection": {
"lcdTimeMode": 4,
"resetScript": "",
"serial": {
"baudrate": 115200,
"communicationTimeout": 3,
"device": "/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0",
"inputBufferSize": 127,
"pingPong": false
}
},
"extruders": [
{
"acceleration": 5000,
"alias": "",
"changeFastDistance": 20,
"changeSlowDistance": 20,
"cooldownPerSecond": 0.5,
"eJerk": 20,
"extrudeSpeed": 2,
"filamentDiameter": 1.75,
"heatupPerSecond": 2,
"lastTemp": 238,
"maxSpeed": 50,
"maxTemp": 260,
"offsetX": 0,
"offsetY": 0,
"retractSpeed": 30,
"temperatures": [
{
"name": "ABS 245",
"temp": 245
},
{
"name": "PLA 210",
"temp": 210
},
{
"name": "PLA 195",
"temp": 195
}
]
}
],
"general": {
"active": true,
"defaultVolumetric": false,
"eepromType": "repetier",
"fan": true,
"fan2": false,
"firmwareName": "Repetier-Firmware",
"heatedBed": true,
"logHistory": true,
"name": "Krieger Klone Macher",
"pauseHandling": 0,
"pauseSeconds": 120,
"printerVariant": "cartesian",
"sdcard": false,
"slug": "Krieger_Klone_Macher",
"softwarePower": true,
"tempUpdateEvery": 1,
"useModelFromSlug": "",
"useOwnModelRepository": true
},
"heatedBed": {
"cooldownPerSecond": 0.02,
"heatupPerSecond": 0.2,
"installed": true,
"lastTemp": 0,
"maxTemp": 120,
"temperatures": [
{
"name": "ABS",
"temp": 115
},
{
"name": "PLA",
"temp": 50
}
]
},
"heatedChamber": {
"installed": false,
"maxTemp": 100
},
"movement": {
"G10Distance": 3,
"G10LongDistance": 40,
"G10Speed": 40,
"G10ZLift": 0,
"G11ExtraDistance": 0,
"G11ExtraLongDistance": 0,
"G11Speed": 20,
"allEndstops": true,
"invertX": false,
"invertY": false,
"invertZ": false,
"maxXYSpeed": 200,
"maxZSpeed": 2,
"movebuffer": 20,
"timeMultiplier": 1,
"xEndstop": true,
"xHome": 0,
"xMax": 200,
"xMin": 0,
"xyJerk": 10,
"xyPrintAcceleration": 1000,
"xySpeed": 100,
"xyTravelAcceleration": 1000,
"yEndstop": true,
"yHome": 0,
"yMax": 200,
"yMin": 0,
"zEndstop": true,
"zHome": 0,
"zJerk": 0.71,
"zMax": 180,
"zMin": 0,
"zPrintAcceleration": 100,
"zSpeed": 2,
"zTravelAcceleration": 100
},
"properties": {},
"quickCommands": [],
"shape": {
"basicShape": {
"color": "#dddddd",
"radius": 100,
"shape": "rectangle",
"x": 0,
"xMax": 200,
"xMin": 0,
"y": 0,
"yMax": 200,
"yMin": 0
},
"gridColor": "#454545",
"gridSpacing": 10,
"marker": []
},
"webcam": {
"dynamicUrl": "",
"method": 0,
"orientation": 0,
"reloadInterval": 3,
"staticUrl": "",
"timelapseBitrate": 1000,
"timelapseFramerate": 30,
"timelapseHeight": 0.1,
"timelapseInterval": 20,
"timelapseLayer": 1,
"timelapseMethod": 0,
"timelapseSelected": 0
}
}```

badvision, 2018-02-11:
Quick note: Not sure how the key being passed around is the printer type instead of its name, but I hotwired the code to genertate the URL correctly and it still crashes when it is parsing the output.  It's something going on when trying to evaluate the numextruders, which is why I included the JSON coming back from RepetierServer for your reference.

VMaxx, 2018-02-12:
Its running stateList to get the numextruders.  Yes the instance name has to match the Config name that is setup in Repetier.    Send me what you get from stateList

B3DTech, 2018-02-13:
So far I was able to get Cura connected to Repetier Host with the 3.2 branch.  I think we're going a little overkill with the 15 decimal place precision on the bed temp though :)  Haven't tried to print with it yet.  Won't be able to get to that for a few days.  

[image]

VMaxx, 2018-02-13:
Great thanks for the reply!

VMaxx, 2018-02-14:
I updated it with some error handling so it won't dump cura if its misconfigured.

VMaxx, 2018-02-16:
I will close this issue if you're happy it's resolved.

B3DTech, 2018-03-01:
I was able to use it just fine with the latest Cura.

closed 2018-03-06 by VMaxx