VMaxx/RepetierIntegration #3: Confused on the configuration

buzzwang, 2018-03-17:
I'm confused on the configuration of the plugin. I understand that the instance name in the plugin must match the name of the printer in Cura. I call my printer Yellowjacket, and have it named that way in both Cura and on the Repetier-Server.

Hostname and port are self explanatory. The host is also named yellowjacket.

What 'Path' is it asking for? I have no idea what to put there.

And which API key is it we should use? The User key or the Global Settings key? I've tried both and can't get anything but an error saying "Repetier on Yellowjacket configuration is invalid". I do have a user name and password setup on the server, and did enter those in the plugin configuration window.

I'm not sure what the error is saying, but I'm guessing it doesn't like the information I provided? I've confirmed the username and password are correct, I've used both API keys without success, and everything is named Yellowjacket, though I did try the server IP address instead of hostname a few times. Regardless, each time I get the same message about an invalid configuration. More complete setup instructions would be nice.

VMaxx, 2018-03-19:
I've attached an image to helpfully show you how it should be configured.  The path you leave as default.
[image]

buzzwang, 2018-03-21:
What you're showing in the mage s what I had entered along with the API key
in the window before this one (tries both keys actually), and ib as never
able to get it to say anything's ng more than the error I listened at the
beginning about it being configured incorrectly. I know my server works as
I also use Repetier Host to run things.

I thought maybe I was just misunderstanding the config process, but based
on the image I had it spot-on So I don't know what the problem is. I'm
planned ng on printing more this evening and I can play with the settings
so more at that time.

<https://user-images.githubusercontent.com/12956626/37628328-fae3cbc6-2ba6-11e8-93bc-2b138debe246.jpg>

VMaxx, 2018-04-03:
Log files are in %appdata%\cura
stderr.log and stdout.log

buzzwang, 2018-04-07:
It’s not happy with something:

 

2018-04-07 08:28:51,510 - INFO - [(16176)-MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device YJ already added

2018-04-07 08:28:51,512 - INFO - [(16176)-MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device YJ already added

2018-04-07 08:28:51,514 - INFO - [(16176)-MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device YJ already added

2018-04-07 08:28:51,534 - DEBUG - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice.connect [352]: Connection with instance YJ with url http://yellowjacket:3344/ started

2018-04-07 08:28:51,577 - DEBUG - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice.connect [352]: Connection with instance YJ with url http://yellowjacket:3344/ started

2018-04-07 08:28:51,688 - INFO - [(16176)-MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device YJ already added

2018-04-07 08:28:51,690 - INFO - [(16176)-MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device YJ already added

2018-04-07 08:28:51,692 - INFO - [(16176)-MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device YJ already added

2018-04-07 08:28:51,695 - INFO - [(16176)-MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device YJ already added

2018-04-07 08:28:51,697 - INFO - [(16176)-MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device YJ already added

2018-04-07 08:28:51,697 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:28:51,701 - WARNING - [(16176)-MainThread] UM.Qt.QtApplication.__onQmlWarning [267]: file:///C:/Program Files/Ultimaker Cura 3.2/resources/qml/Preferences/MachinesPage.qml:176:17: Unable to assign [undefined] to QString

2018-04-07 08:28:51,733 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:28:53,681 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:28:55,615 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:28:57,706 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:28:59,674 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:01,694 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:03,718 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:05,723 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:07,691 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:09,732 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:11,701 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:13,706 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:15,696 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:17,717 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:19,706 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:21,710 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:23,730 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:25,689 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:27,655 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:29,696 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:29:31,682 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

[... 30 lines cut ...]
[... cut ...]
2018-04-07 08:30:21,782 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:30:23,785 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:30:25,804 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

2018-04-07 08:30:27,773 - WARNING - [(16176)-MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.

VMaxx, 2018-04-11:
Somewhere you have something set to YJ, in the Cura config and not yellowjacket.  The instance name in Cura has to match the name in Repetier.

VMaxx, 2018-04-16:
Right, so your printer in Cura should be named yellowjacket not yj.

VMaxx, 2018-04-17:
Yes it appears to be trying to send to YJ, and in repetier it's yellowjacket.  Either way so long as they match, it should work.  Though deleting it in Cura is a lot quicker and easier than renaming it in Repetier.

buzzwang, 2018-04-17:
Figured it out after reading through the API docs for Repetier.

 

Repetier wants to use the printer ‘slug’ which is a shortened or abbreviated version of the printer name in Repetier Server.

 

In order for that to work, the name of the printer in Cura 3.2.1 must match the slug name used by Repetier Server. Once I sorted that bit out, it connected right away.

 

For reference, in a web browser:

 

http://192.168.1.21:3344/printer/api/Yellowjacket?a=getPrinterConfig <http://192.168.1.21:3344/printer/api/Yellowjacket?a=getPrinterConfig&apikey=%3ckey_omitted_for_security> &apikey=<key_omitted_for_security>

… would get…

{"error":"Unknown printer"}

 

 

But, using:

http://192.168.1.21:3344/printer/api/Yellow?a=getPrinterConfig <http://192.168.1.21:3344/printer/api/Yellow?a=getPrinterConfig&apikey=6227e17c-d743-468d-9ab5-a39253a73b45> &apikey=6227e17c-d743-468d-9ab5-a39253a73b45

…would get…

A whole slew of JSON code about my printer.

 

So, then I just had to figure out where in the URL the name Yellowjacket was coming from. Trial and error revealed that it was the actual printer name as setup in Cura. Then I renamed the printer to ‘Yellow’, which matches the slug name that Repetier Server uses, and voila! It works.

 

I found in the Repetier API docs how to get/find the slug name, don’t recall the command/URL now but it was listed towards the beginning of the API documentation.

closed 2018-06-25 by VMaxx