VMaxx/RepetierIntegration #6: Cura 3.3 closes on clicking connect

Beliar-666, 2018-06-02:
Hi. 
Is there any way to connect Cura 3.3 with this Plugin? Cura always closes when i try to connect to repetier-server.

Beliar-666, 2018-06-02:
`2018-06-02 16:41:39,815 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///I:/Ultimaker Cura 3.3/resources/qml/SidebarHeader.qml:20: TypeError: Cannot read property 'length' of undefined
2018-06-02 16:41:39,818 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [277]: file:///I:/Ultimaker Cura 3.3/resources/qml/Preferences/MachinesPage.qml:162: TypeError: Cannot read property 'connectionText' of null
2018-06-02 16:41:39,842 - DEBUG - [MainThread] RepetierPlugin.RepetierOutputDevice.connect [352]: Connection with instance A8 with url http://192.168.178.60:80/ started
2018-06-02 16:41:39,844 - WARNING - [MainThread] UM.Decorators.deprecated_function [19]: <function PrinterOutputDevice.setAcceptsCommands at 0x000001EAC85EBAE8> is deprecated (since 3.2): Please use the protected function instead
2018-06-02 16:41:39,894 - DEBUG - [MainThread] RepetierPlugin.RepetierOutputDevice.connect [352]: Connection with instance A8 with url http://192.168.178.60:80/ started
2018-06-02 16:41:39,897 - WARNING - [MainThread] UM.Decorators.deprecated_function [19]: <function PrinterOutputDevice.setAcceptsCommands at 0x000001EAC85EBAE8> is deprecated (since 3.2): Please use the protected function instead
2018-06-02 16:41:39,942 - WARNING - [MainThread] UM.Decorators.deprecated_function [19]: <function PrinterOutputDevice.setAcceptsCommands at 0x000001EAC85EBAE8> is deprecated (since 3.2): Please use the protected function instead
2018-06-02 16:41:39,944 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device A8 already added
2018-06-02 16:41:39,947 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device A8 already added
2018-06-02 16:41:39,947 - WARNING - [MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.
2018-06-02 16:41:39,952 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-06-02 16:41:39,969 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-06-02 16:41:39,972 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Users\MatzeB\AppData\Roaming\cura\3.3\plugins\RepetierPlugin\RepetierOutputDevice.py", line 656, in _onRequestFinished
2018-06-02 16:41:39,974 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     if "numExtruder" in json_data[self._key]:
2018-06-02 16:41:39,976 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: KeyError: 'A8'
2018-06-02 16:41:39,978 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: 
2018-06-02 16:41:39,981 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: During handling of the above exception, another exception occurred:
2018-06-02 16:41:39,983 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-06-02 16:41:39,985 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Users\MatzeB\AppData\Roaming\cura\3.3\plugins\RepetierPlugin\RepetierOutputDevice.py", line 687, in _onRequestFinished
2018-06-02 16:41:39,988 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self._updateJobState("offline")
2018-06-02 16:41:39,990 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: AttributeError: 'RepetierOutputDevice' object has no attribute '_updateJobState'
2018-06-02 16:41:39,992 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-06-02 16:41:39,995 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-06-02 16:41:39,997 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Users\MatzeB\AppData\Roaming\cura\3.3\plugins\RepetierPlugin\RepetierOutputDevice.py", line 713, in _onRequestFinished
2018-06-02 16:41:39,999 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     if "job" in json_data[0]:
2018-06-02 16:41:40,001 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: KeyError: 0
2018-06-02 16:41:40,003 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: 
2018-06-02 16:41:40,006 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: During handling of the above exception, another exception occurred:
2018-06-02 16:41:40,007 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-06-02 16:41:40,009 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Users\MatzeB\AppData\Roaming\cura\3.3\plugins\RepetierPlugin\RepetierOutputDevice.py", line 755, in _onRequestFinished
2018-06-02 16:41:40,011 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self._updateJobState("offline")
2018-06-02 16:41:40,013 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: AttributeError: 'RepetierOutputDevice' object has no attribute '_updateJobState'
2018-06-02 16:41:40,014 - WARNING - [MainThread] RepetierPlugin.RepetierOutputDevice._onRequestFinished [685]: Received invalid JSON from Repetier instance.
2018-06-02 16:41:40,016 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-06-02 16:41:40,018 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
[... cut ...]
2018-06-02 16:41:40,030 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-06-02 16:41:40,031 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Users\MatzeB\AppData\Roaming\cura\3.3\plugins\RepetierPlugin\RepetierOutputDevice.py", line 687, in _onRequestFinished
2018-06-02 16:41:40,033 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self._updateJobState("offline")
2018-06-02 16:41:40,035 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: AttributeError: 'RepetierOutputDevice' object has no attribute '_updateJobState'
`

VMaxx, 2018-06-02:
Yes, unfortunately there's currently a bug in Cura 3.3 that prevents me from releasing the plugin for the 3.3 version.  The bug in question is if you use a webcam, it doesn't work in Cura (or sometimes it does for whatever unknown reason).  I've been waiting for a resolution to it before releasing the RepetierPlugin because I figured it could cause a lot of grief otherwise.  Until then, I have rolled back to 3.2.  The 3.2 plugin won't work for 3.3 as they've moved some of the structure around the calls that worked in 3.2 won't work at all in 3.3.

So for now I'd recommend rolling back to 3.2 if you're using this plugin.

Appesteijn, 2018-06-12:
Is the bug fixed in Cura 3.4.0-BETA ?

VMaxx, 2018-06-21:
I've posted a version for 3.3 now.  I updated the master branch to match 3.3

closed 2018-06-21 by VMaxx