VMaxx/RepetierIntegration #9: Possibility to only Upload the gCode to Cura

Beliar-666, 2018-07-15:
Is it possible to change the behaviour of the Plugin? The Programmers of Repetier say it is only one url to change.
The Plugin actually uploads the GCode to the Printqueue. So after printing it, it dissapears from Repetier-server. I think it is more comfortable to upload the gCode only to the filelist, so i can print it every time i want. 
Here is the Answer from Repetier:
[https://forum.repetier.com/discussion/comment/22456#Comment_22456](url)

VMaxx, 2018-07-17:
I'll see what I can do.  I've just been cloning what the Octoprint plugin does, then swapping out the calls to talk to Repetier.

VMaxx, 2018-07-19:
I've uploaded an update that when you configure the plugin there's a checkbox for "Automatically start print job after uploading" it was previously ignored.  Now when it is unchecked, it will store in Repetier instead of printing immediately.

Beliar-666, 2018-07-20:
i´ll give it a try.

VMaxx, 2018-07-22:
awesome!

closed 2018-07-22 by VMaxx

userosos, 2019-03-18:
@VMaxx can you add "Automatically start print job after uploading". And add gcode file into special group "Cura", for example. And start print after uploading. It will be good. Without "Automatically start print job after uploading" very uncomfortable.