VMaxx/RepetierIntegration #11: Cura plugin not working

jeffgar2000, 2018-08-15:
I am using Cura 3.4.1 and Repetier server version  0.90.0 .

My repetier host connects and works with the server just fine.  I installed the Cura plugin and it allows me to configure the key IP address and port but when I try to connect it just crashes and closes cura.

The cura.log was too big to attach so I attached a file (cura2.log) which is ithe bottom portion of the log that should show the problem.

Thanks

[cura2.log](https://github.com/VMaxx/RepetierPlugin/files/2291921/cura2.log)

VMaxx, 2018-08-16:
I suspect its the name, in one place you have the printer name as "Jeffs I3" and in another its just "JeffsI3"  So the stub (ie config in the repetier web) doesn't match what you have in the plugin for instance name.  I'd try removing the space in the repetier web and see if that fixes it.

jeffgar2000, 2018-08-16:
I did have one of those wrong when you select the printer you want to connect to repetier host.  I changed all occurrences in repetier and Cura to JeffsI3 with no space.  Repetier host to server works fine.  Cura still crashes.

Here is the applicable portion of the log I think.
2018-07-07 17:47:55,625 - DEBUG - [MainThread] cura.AutoSave._onTimeout [49]: Autosaving preferences, instances and profiles
2018-07-07 17:48:13,438 - DEBUG - [MainThread] RepetierPlugin-master.DiscoverRepetierAction.apiKey [188]: APIKEY read None
2018-07-07 17:48:13,517 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-07-07 17:48:13,517 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-07-07 17:48:13,532 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "D:\Program Files\Ultimaker Cura 3.4\plugins\RepetierPlugin\RepetierPlugin-master\DiscoverRepetierAction.py", line 72, in startDiscovery
2018-07-07 17:48:13,532 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self._network_plugin.addInstanceSignal.connect(self._onInstanceDiscovery)
2018-07-07 17:48:13,548 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: AttributeError: 'NoneType' object has no attribute 'addInstanceSignal'
2018-07-07 17:48:23,531 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin FileLogger
2018-07-07 17:48:23,719 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin FirmwareUpdateChecker

VMaxx, 2018-08-18:
It doesn't appear to have picked up the apikey.

jeffgar2000, 2018-08-18:
Yes I saw that too but I definitely entered the API key.  The same one I use from repetier host to connect.  I even entered it, saved it closed the program, reran and made sure it was still there before trying to run again but it still crashed.

jeffgar2000, 2018-08-19:
[Repetier]
manual_instances = {"JeffsI3": {"path": "/", "useHttps": false, "password": "", "address": "192.168.0.40", "port": 3344, "userName": ""}}

What's it missing?

I really appreciate the help.  I really like working in cura but need the repetier server.

VMaxx, 2018-08-19:
[Repetier]
base_url = http://192.168.1.250:3344/
printer = Tarantula
manual_instances = {"Tarantula": {"userName": "", "password": "", "address": "192.168.1.250", "path": "/", "port": 3344, "useHttps": false}}
api_key = put your api key here

jeffgar2000, 2018-08-19:
OK I tried it in the format you have and I also tried it like:

[Repetier]
manual_instances = {"JeffsI3": {"path": "/", "useHttps": false, "password": "", "address": "192.168.0.40", "port": 3344, "userName": "", "api_key": "1ae32bfd-a845-4a07-829e-7df51c3f5bed"}}

Sill crashes out.  It was kind of obvious it would not work because both times when I went into the menu to connect, the API key was blank although I had already added it to the file.

VMaxx, 2018-08-20:
Try the following for you:
`[Repetier]
base_url = http://192.168.0.40:3344/
manual_instances = {"JeffsI3": {"path": "/", "useHttps": false, "password": "", "address": "192.168.0.40", "port": 3344, "userName": ""}}
api_key = 1ae32bfd-a845-4a07-829e-7df51c3f5bed`

jeffgar2000, 2018-08-21:
Nope.  Still crashes.

I copied your exact example and pasted it in so no chance of mistype.

VMaxx, 2018-08-21:
ok send me the log

VMaxx, 2018-08-21:
also check the %Appdata%\cura\3.4\machine_instance there should be a cfg file there.

VMaxx, 2018-08-21:
some people have just deleted it from the repetier plugin then re-added it and it works.

jeffgar2000, 2018-08-21:
I tried reloading a few different times before I posted the message for help.  I even reloaded the Repetier server in case the issue was on that end.

That path does have a file named Jeffs+I3.global.cfg

Below are the contents:

[general]
version = 4
name = JeffsI3
id = Jeffs I3

[metadata]
setting_version = 4
type = machine
repetier_auto_print = false

[containers]
0 = Jeffs I3_user
1 = empty_quality_changes
2 = normal
3 = empty_material
4 = empty_variant
5 = Jeffs I3_settings
6 = prusa_i3

jeffgar2000, 2018-08-21:
The log is too big to attach but here is the last snippet that shows the error when I try to connect:

2018-08-20 20:43:50,962 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_abs_175_cartesio_0.8mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:50,972 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_petg_175_cartesio_0.25mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:50,982 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding dsm_novamid1070_175_cartesio_0.25mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:50,992 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding dsm_arnitel2045_175_cartesio_0.25mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,002 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_hips_175_cartesio_0.4mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,012 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_nylon_175_cartesio_0.25mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,012 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_pva_175_cartesio_0.8mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,022 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_abs_175_cartesio_0.4mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,022 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_pla_175_cartesio_0.25mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,032 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_pva_175_cartesio_0.25mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,043 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_pc_175_cartesio_0.8mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,043 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_pla_175_cartesio_0.4mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,060 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_pla_175_cartesio_0.8mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,067 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_hips_175_cartesio_0.25mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,076 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_pva_175_cartesio_0.4mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,082 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding dsm_novamid1070_175_cartesio_0.8mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,083 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_abs_175_cartesio_0.25mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,093 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_nylon_175_cartesio_0.4mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,093 - WARNING - [MainThread] cura.Machines.MaterialManager.initialize [230]: Not adding generic_petg_175_cartesio_0.8mm_thermoplastic_extruder to the material manager because the variant does not exist.
2018-08-20 20:43:51,103 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1163]: Updating quality/quality_changes due to material change
2018-08-20 20:43:51,113 - DEBUG - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1170]: Current quality type = [normal]
2018-08-20 20:43:51,123 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1184]: Current available quality type [normal] is available, applying changes.
2018-08-20 20:43:51,153 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-08-20 20:43:51,477 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [403]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.020000457763671875 seconds. 36 input vertices. 36 output vertices.
2018-08-20 20:43:53,330 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin._createView [256]: Creating post processing plugin view.
[... cut ...]
2018-08-20 20:45:00,126 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device JeffsI3 already added
2018-08-20 20:45:00,131 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device JeffsI3 already added
2018-08-20 20:45:00,133 - WARNING - [MainThread] RepetierConnection.RepetierOutputDevice._onRequestFinished [605]: Received invalid JSON from Repetier instance.
2018-08-20 20:45:00,135 - WARNING - [MainThread] RepetierConnection.RepetierOutputDevice._onRequestFinished [605]: Received invalid JSON from Repetier instance.

VMaxx, 2018-08-22:
Yeah I'm thinking its still related to the name.
You might be better off deleting the printer out of cura and restarting with no spaces.  I'll look and see if I can code some way to deal with spaces in the mean time.
"id = Jeffs I3"

jeffgar2000, 2018-08-23:
I renamed to "Prusa", restarted everything, Removed the old printer in Cura and created a new one name Prusa.  I generated a new API key too.  Click to connect and it shut completely down again.

OK at this point I'm going to go in and uninstall anything Cura related and start all over.  I have directories for multiple versions stranded from 2.x up to 3.4.1.  Once I get everything removed then I will go reload and try again.  I don't really think anything is wrong with the plugin so much as maybe I just have too much related baggage in my system.  

Give me till Monday to reload and I'l report back.

VMaxx, 2018-08-23:
Ok, the reason I was thinking its something with the name is because its dying on the line that reads the settings in general, not really specific to the plugin which is unexpected.

VMaxx, 2018-08-23:
Check in  C:\Users\username\AppData\Local\VirtualStore and see if cura stuff is in there.  Just incase Windows decided to stick things in there due to rights issues.  It stuffs up things sometimes.

jeffgar2000, 2018-08-23:
OK so I see NO cura related files in that directory.   I will reload everything with a completely different name and see what I get.

I really appreciate your help.

HamOP, 2018-08-25:
I just had a very similar issue and finally found that the reason was that Repetier Server doesn't necessarily name the printer internally as you name it in the server. It creates a slug that is not changed even if you rename the printer. I had my printer named "Anycubic i3 Ultrabase" and it didn't work, also changing the name on both sides to "i3Ultra" or something like this didn't help. Using the command "http://server-ip-here:3344/printer/info", I finally found out that the printer slug actually was "Anycubic_i3_Ultrabase" and didn't change even when I renamed the printer. So, for your config, I guess that if you put "Jeffs_I3" in as the instance name in the plugin, it should work. As your original name was "Jeffs I3", I guess that Repetier Server automatically changed the space to an underscore for the slug as it did for me.

BTW, I found that the length of the instance name field in Cura is limited, so it didn't take my lengthy printer names. After changing to the full name of the instance manually in cura.cfg, it now works perfectly.

jeffgar2000, 2018-08-27:
Wow OK that makes sense.  I have it mostly realoaded and I will definitely check this as soon as I get it back up.

THANK YOU for posting that

jperrin72, 2018-08-27:
FYI got the same issue and proposed a code fix in a separate thread :)

VMaxx, 2018-08-27:
awesome

closed 2018-08-28 by VMaxx