VMaxx/RepetierIntegration #12: Crash when connecting

jperrin72, 2018-08-25:
Hi there,

I got an issue when using the "Connect Repetier" button which was crashing Cura interface.
I found out that exception was raised from DiscoverRepetierAction.py.
Comparing with plugin for  OctoPrintConnect I changes the line #72 from:

self._network_plugin = Application.getInstance().getOutputDeviceManager().getOutputDevicePlugin("RepetierPlugin")

to:

self._network_plugin = Application.getInstance().getOutputDeviceManager().getOutputDevicePlugin(self._plugin_id)

et voila!

I thought you may want to integrate this fix into the main branch :)

(modified file attached)

Thanks

Jerome

[DiscoverRepetierAction.txt](https://github.com/VMaxx/RepetierPlugin/files/2320976/DiscoverRepetierAction.txt)

VMaxx, 2018-08-27:
Great I'll have to let the Octoprint guy know, his has similar code.

jperrin72, 2018-08-28:
actually I just replicated this code from Octoprint (DiscoverOctoprintAction.py)

VMaxx, 2018-08-28:
I went over to tell them and saw they had changed it.  He must have changed it since I copied it.

closed 2018-08-30 by VMaxx