VMaxx/RepetierIntegration #13: Cura 3.4.1 Crashes

unlimitopen, 2018-09-17:
hello together,
i have installed the repetier plugin and after i insert the api key, cura will complete crash.
`2018-09-17 19:54:45,922 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-09-17 19:54:45,924 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-09-17 19:54:45,926 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "C:\Program Files\Ultimaker Cura 3.4\plugins\RepetierPlugin\DiscoverRepetierAction.py", line 329, in _onRequestFinished
2018-09-17 19:54:45,928 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     if "webcams" in json_data and "dynamicUrl" in json_data["webcams"][0]:
2018-09-17 19:54:45,930 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: IndexError: list index out of range`

After this information from log file, i changed the list in 
`DiscoverRepetierAction.py` and in `RepetierOutputDevice.py` in the dictionary `json_data`.

After this change on the plugin files... it works fine.

VMaxx, 2018-09-20:
Do you have a webcam?

VMaxx, 2018-09-20:
Posted a fix for this it will check to see if you have webcams, if not it skips the logic.

closed 2018-09-27 by VMaxx