VMaxx/RepetierIntegration #17: Repetier Plugin

Ellecross, 2018-10-30:
Hi! 

I'm currently testing all plugins with the new Cura 3.6 Beta - which is planned to be released this week, if not today. 

Since at the moment, RepetierPlugin is compatible only for 3.5, can you also make repetierplugin compatible as well for Cura 3.6? 

Thanks!

VMaxx, 2018-11-01:
Ok, I've not gotten an e-mail on it.

VMaxx, 2018-11-01:
Looks like they changed the camera code, so I'll have to adjust.

VMaxx, 2018-11-03:
I updated it on github.  I tried looking for an update or new package in the contribute site for ultimaker and it wasn't clear on how to upload an update without changing the name.  I may just have to do that.

VMaxx, 2018-11-03:
I've e-mailed it to them with the problem they've got with the site.  It won't even accept the package I uploaded last time so there's definitely an issue there.

VMaxx, 2018-11-05:
I got it fixed, there was a context menu I didn't see.  It should be in review now.

closed 2018-11-05 by VMaxx

Ellecross, 2018-11-05:
Thank you! Our product owner is now reviewing it!