VMaxx/RepetierIntegration #18: Repetier for official launch of Cura

Ellecross, 2018-11-09:
Hi, sorry to come again with this issue. 

But I'm testing now with the new 3.6 that will be official build (no longer beta). 

I tried installing the 3.5.2 version of the Repetier plugin again and I get the following errors on launch. 

Is this similar to what happened previously? 

2018-11-09 10:31:54,458 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [587]: Could not find plugin RepetierIntegration
2018-11-09 10:31:54,469 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Import error loading module RepetierIntegration
2018-11-09 10:31:54,473 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2018-11-09 10:31:54,475 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\3.6\build\inst\lib\python3.5\site-packages\UM\PluginRegistry.py", line 547, in _findPlugin
2018-11-09 10:31:54,478 - ERROR - [MainThread] UM.Logger.logException [88]:   File "imp.py", line 244, in load_module
2018-11-09 10:31:54,480 - ERROR - [MainThread] UM.Logger.logException [88]:   File "imp.py", line 216, in load_package
2018-11-09 10:31:54,483 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 693, in _load
2018-11-09 10:31:54,485 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2018-11-09 10:31:54,487 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap_external>", line 665, in exec_module
2018-11-09 10:31:54,491 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2018-11-09 10:31:54,495 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\t.delacruz\AppData\Roaming\cura\3.6\plugins\RepetierIntegration\RepetierIntegration\__init__.py", line 5, in <module>
2018-11-09 10:31:54,497 - ERROR - [MainThread] UM.Logger.logException [88]:     from . import RepetierOutputDevicePlugin
2018-11-09 10:31:54,500 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\t.delacruz\AppData\Roaming\cura\3.6\plugins\RepetierIntegration\RepetierIntegration\RepetierOutputDevicePlugin.py", line 2, in <module>
2018-11-09 10:31:54,502 - ERROR - [MainThread] UM.Logger.logException [88]:     from . import RepetierOutputDevice
2018-11-09 10:31:54,507 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\t.delacruz\AppData\Roaming\cura\3.6\plugins\RepetierIntegration\RepetierIntegration\RepetierOutputDevice.py", line 13, in <module>
2018-11-09 10:31:54,511 - ERROR - [MainThread] UM.Logger.logException [88]:     from cura.PrinterOutput.NetworkCamera import NetworkCamera
2018-11-09 10:31:54,514 - ERROR - [MainThread] UM.Logger.logException [88]: ImportError: No module named 'cura.PrinterOutput.NetworkCamera'

VMaxx, 2018-11-11:
Maybe folder name related WARNING - [MainThread] UM.PluginRegistry._populateMetaData [587]: Could not find plugin RepetierIntegration  The Folder name is still RepetierPlugin I haven't changed its folder name to RepetierIntegration yet.

VMaxx, 2018-11-11:
Ok I'm going to resubmit it under the new name.  I'll have to change it here

closed 2018-12-06 by VMaxx