VMaxx/RepetierIntegration #19: camera nowork

userosos, 2019-02-01:
Hi! Camera not work in cura 3.6.
[image]
Api-key from user (admin).

userosos, 2019-02-01:
Now i can see that all not work. I can not send to print too. I can not goto home and set temperatures.
[image]

VMaxx, 2019-02-03:
Do you have security setup?  (ie. a username/password)

userosos, 2019-02-03:
Yes. Camera don't work now in cura.

VMaxx, 2019-02-04:
if you have security turned on, you'll have to click the advanced button and type your userid/password in so that cura can use it to connect.

userosos, 2019-02-05:
I have security turned on. But now camera not work only.
[image]

userosos, 2019-02-06:
I can see it:
[image]

VMaxx, 2019-02-11:
Try changing the camera to MPEG stream https://user-images.githubusercontent.com/12956626/42852880-a54c0d1c-89f8-11e8-8541-bf9d691cbae4.jpg

userosos, 2019-02-17:
@VMaxx it is not work for me. :(

VMaxx, 2019-02-25:
The port should stay 3344 most likely.  I don't have any extra buttons to rotate the screen.  I'll have to add some.  The Flip Y I added mainly for myself.  Do you need to flip X or do a 90 degree rotation?

userosos, 2019-02-25:
> 
> 
> The port should stay 3344 most likely. I don't have any extra buttons to rotate the screen. I'll have to add some. The Flip Y I added mainly for myself. Do you need to flip X or do a 90 degree rotation?

I need all of them, please. Can you read the setting from repieter-server?

VMaxx, 2019-03-05:
Ok I added the extra options in there.

VMaxx, 2019-03-18:
I'm assuming this fixed you're issues.

closed 2019-03-18 by VMaxx