VMaxx/RepetierIntegration #20: Camera option disabled

JerksonS, 2019-03-03:
I have read through issue #19 but it didnt resolve my issue here.

I can upload files and control the printer from Cura using the plugin, so I know my settings are correct.
Also, contrary to the instructions, the instance name that needs to be typed into the plugin is not necessarily the display name of the printer in the web UI, but the name from the URL. If you have changed the display name of the printer since adding, the display name will not work. (picture 3 demonstrates the difference. Display Name "test1" but url shows the original name "test" which must be used in the plugin config)

However, the camera option remains disabled. I even have changed it to MJPEG as suggested in the other issue.
[image]
[image]
[image]

VMaxx, 2019-03-05:
The program uses the instance name for getting its config.
So in your case it will query:
http://10.0.1.111:3344/printer/api/test?a=getPrinterConfig
and not get the correct config because it should be querying:
http://10.0.1.111:3344/printer/api/test1?a=getPrinterConfig
so the name of the printer and the instance in the plugin config does have to match, this is likely why its not detecting the webcam.
you can check %appdata%\Roaming\cura\stderr.log for errors.

VMaxx, 2019-03-05:
You may be able to rename the cfg file in %appdata%\Roaming\cura\3.6\machine_instances\test.global.cfg
and edit it so that the insides are matching.

VMaxx, 2019-03-18:
I am uncertain if it will be better to add a configuration line to be able to have the instance name and repetier printer name different as it may confuse more people than help.  Since the requirement is in the instructions I'm not going to consider not following instructions a bug.

closed 2019-03-18 by VMaxx