VMaxx/RepetierIntegration #21: Lost connection to the printer

Krisshellman1, 2019-03-09:
I have it set up on Cura 3.6 and repetier server 0.90.7--But when I tell it to print--It says sending data just fine--But when it gets to storing data, it stops and says connection to printer lost.

Note: The link to open repetier works fine once the upload fails.
**What does this mean?**

[image]\

[image]

VMaxx, 2019-03-18:
It would appear your config isn't correct.  The sending data to Repetier piece, you can see it already says Lost connection with the printer.    In the config part there is a connect button, click to connect it should then say ready in the lower right instead of lost connection.

Krisshellman1, 2019-03-21:
Thanks! I fixed it--The problem was that I named it the same as I set the name in repetier server 'Mondel'
But infact the printed actual name was changed to 'mendel1' for some reason.

I cant believe this plugin works, this changes everything!

VMaxx, 2019-03-22:
Great glad it's working for you!

closed 2019-03-22 by VMaxx