VMaxx/RepetierIntegration #22: Turn off bed heater and extruder  heater unexpected

shaddow501, 2019-03-19:
Hello

If I press in cura 3.6 "Print with Repetier" it starts printing and after a while I notice that the bed heater goes off and also the extruder heater goes off.

If I save as gcode and upload to the repetier server everything goes well. heaters working till the end of the print.

VMaxx, 2019-03-26:
You may need to update repetier, there was some specific temperature related changes recently.

shaddow501, 2019-03-27:
I am already on the latest version of repetier...

VMaxx, 2019-03-28:
Ok I'll check with my printer.  I usually preheat and haven't had this happen to me.  It may be printer dependent.

VMaxx, 2019-03-29:
I just tested this, and it worked without dropping either one.  You may have another issue.

closed 2019-06-08 by VMaxx