VMaxx/RepetierIntegration #23: Cura 4.0 support

betavr, 2019-03-20:
Cura 4.0 final was just released. Please update the plugin to support it!

Related issue: https://github.com/Ultimaker/Cura/issues/5459

betavr, 2019-03-22:
I installed the plugin manually but can't get it to work. Every 2 seconds I get these messages in cura.log:

```
2019-03-22 11:49:17,123 - WARNING - [MainThread] UM.Logger.logException [84]: Exception: something went wrong with callback
2019-03-22 11:49:17,124 - WARNING - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2019-03-22 11:49:17,126 - WARNING - [MainThread] UM.Logger.logException [88]:   File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\PrinterOutput\NetworkedPrinterOutputDevice.py", line 346, in __handleOnFinished
2019-03-22 11:49:17,127 - WARNING - [MainThread] UM.Logger.logException [88]:   File "C:\Users\beta\AppData\Roaming\cura\4.0\plugins\RepetierIntegration\RepetierIntegration\RepetierOutputDevice.py", line 650, in _onRequestFinished
2019-03-22 11:49:17,128 - WARNING - [MainThread] UM.Logger.logException [88]:     if "job" in json_data[0]:
2019-03-22 11:49:17,130 - WARNING - [MainThread] UM.Logger.logException [88]: KeyError: 0
2019-03-22 11:49:17,130 - WARNING - [MainThread] RepetierIntegration.RepetierOutputDevice._onRequestFinished [595]: Received invalid JSON from Repetier instance.
2019-03-22 11:49:19,134 - WARNING - [MainThread] UM.Logger.logException [84]: Exception: something went wrong with callback
2019-03-22 11:49:19,135 - WARNING - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2019-03-22 11:49:19,137 - WARNING - [MainThread] UM.Logger.logException [88]:   File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\PrinterOutput\NetworkedPrinterOutputDevice.py", line 346, in __handleOnFinished
2019-03-22 11:49:19,138 - WARNING - [MainThread] UM.Logger.logException [88]:   File "C:\Users\beta\AppData\Roaming\cura\4.0\plugins\RepetierIntegration\RepetierIntegration\RepetierOutputDevice.py", line 650, in _onRequestFinished
2019-03-22 11:49:19,140 - WARNING - [MainThread] UM.Logger.logException [88]:     if "job" in json_data[0]:
2019-03-22 11:49:19,141 - WARNING - [MainThread] UM.Logger.logException [88]: KeyError: 0
2019-03-22 11:49:19,142 - WARNING - [MainThread] RepetierIntegration.RepetierOutputDevice._onRequestFinished [595]: Received invalid JSON from Repetier instance.
```

I also get other errors depending on what I change in the settings or in repetier. I must say the UI is very confusing. The printer's repetier id should be editable directly by the user, independently of what the printer is called in Cura. I suspect this is the main problem, but I'm unsure at this point because having Cura start with a clean plate is not easy - it always imports configuration from my other Cura installs.

Please tell me how I can help you diagnosing this problem.

VMaxx, 2019-03-24:
The instance name in the plug-in has to match the name you gave it in repetier.  That's the most common issue.  I guess I will have to break down and add a way for them to be different.. but for now delete the printer from the plug-in and make a new one and match the name

userosos, 2019-03-26:
I can not see the plugin in marketplace of cura. Are you planning to add the plug in marketplace?
[image]

VMaxx, 2019-03-26:
The plugin is still under review.  Until they approve it, it won't publish.

betavr, 2019-03-28:
Ok, I managed to get it to work by deleting the folder AppData\Roaming\cura\4.0, and let cura import the previous configuration again. I don't why it didn't work earlier. Closing this for now.

closed 2019-03-28 by betavr