VMaxx/RepetierIntegration #24: When you add an instance with a printer and try to edit it it crashes

Sangoku, 2019-03-24:
When i add a rapetier instance with name, lets say delta, to a config it adds it without problems. WHen the ip or the key changes you pretty much can only remove all of the plugins data only to get it to work again because when you try to add or edit the connection the whole cura crashes. 

I even went so far and changed the printer connection name to make it work again. 
How can i fix such behavior? 
Where are the connections stored?

VMaxx, 2019-03-25:
2 main config files 
%AppData%\Roaming\cura\4.0\cura.cfg
%AppData%\Roaming\cura\4.0\machine_instances\{name you gaveit}.global.cfg

VMaxx, 2019-03-28:
I'm guessing that you have your repetier set to dhcp.   I'd set it to a fixed ip so you won't have to keep changing it.  Otherwise when you load cura the plugin tries to reconnect to the old IP and crashes.  I'll see what I can do to add some error handling for that.

llambkin, 2019-04-03:
I am having the exact same issue with my Cura 3.6, and 4.0 (yes, I cloned the correct branch for the version). It happens when adding the machine to the system, and the printer will not connect, and when going back to edit the instance, or if I press refresh, it crashes with the following error. 

Here is my Error Traceback:
Traceback (most recent call last):
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\DiscoverRepetierAction.py", line 112, in setManualInstance
    self._network_plugin.addManualInstance(name, address, port, path, useHttps, userName, password)
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\RepetierOutputDevicePlugin.py", line 104, in addManualInstance
    self.addInstance(name, address, port, properties)
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\RepetierOutputDevicePlugin.py", line 148, in addInstance
    instance.setApiKey(api_key)
NameError: name 'api_key' is not defined

Logs:
Thread 0x00000a94 (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 169 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000704 (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 161 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000273c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002490 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000c58 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002da8 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001e54 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\UM3NetworkPrinting\src\UM3OutputDevicePlugin.py", line 344 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002cd4 (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\zeroconf.py", line 1725 in wait
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\zeroconf.py", line 1326 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002a0c (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\zeroconf.py", line 1725 in wait
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\zeroconf.py", line 1190 in run
  File "threading.py", line 914 in _bootstrap_inner
[... 70 lines cut ...]
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00002d3c (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 306 in _logInfoWidget
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 156 in _createDialog
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 77 in __init__
  File "X:\4.0\build\inst\bin\cura_app.py", line 98 in exceptHook
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 324 in exec_
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 761 in run
  File "X:\4.0\build\inst\bin\cura_app.py", line 132 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

VMaxx, 2019-04-05:
It isn't seeing your apikey try re-entering it in the config

File "", line 634 in _load_backward_compatible
File "", line 664 in _load_unlocked
File "", line 958 in _find_and_load_unlocked
File "", line 969 in _find_and_load
File "", line 634 in _load_backward_compatible
File "", line 664 in _load_unlocked
File "", line 958 in _find_and_load_unlocked
File "", line 969 in _find_and_load

VMaxx, 2019-04-09:
i added another update to try.

llambkin, 2019-04-17:
Thankyou for your reply, sorry it has taken me so long to reply. I will try your update when I have to reinstall Cura next time. I have got it working with your advice, I added the key into the config files. It was a pain in the butt, but worth it! Being able to "just click print" is a much needed feature.

closed 2019-06-07 by VMaxx