VMaxx/RepetierIntegration #25: No upload, again

PPPBerlin, 2019-04-03:
Cura 4.0 Repetier Server Pro 0.91.2

Afer I slice the model and try to upload the gcode, it connects, try to save the file, but the progress bar didn't move and nothing changed on the Repetier Server.

This is from the cura.log and apperars a few times.

`2019-04-03 09:17:29,115 - WARNING - [MainThread] RepetierIntegration.RepetierOutputDevice._onRequestFinished [595]: Received invalid JSON from Repetier instance.
2019-04-03 09:17:29,119 - WARNING - [MainThread] UM.Logger.logException [84]: Exception: something went wrong with callback
2019-04-03 09:17:29,123 - WARNING - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2019-04-03 09:17:29,125 - WARNING - [MainThread] UM.Logger.logException [88]:   File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\PrinterOutput\NetworkedPrinterOutputDevice.py", line 346, in __handleOnFinished
2019-04-03 09:17:29,127 - WARNING - [MainThread] UM.Logger.logException [88]:   File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\RepetierOutputDevice.py", line 650, in _onRequestFinished
2019-04-03 09:17:29,129 - WARNING - [MainThread] UM.Logger.logException [88]:     if "job" in json_data[0]:
2019-04-03 09:17:29,131 - WARNING - [MainThread] UM.Logger.logException [88]: KeyError: 0`

PPPBerlin, 2019-04-03:
And yes, i checked the other Issues with Server Name etc.

VMaxx, 2019-04-03:
its requesting to see the job but no job is being returned.  Do you have it set to print immediately or store?

VMaxx, 2019-04-03:
the line "Received invalid JSON from Repetier instance." is when python can't decode what it got from the printer.  Basically your repetier device returned bad json that can't be decoded.  It could be a network error or something.  It might be something with the naming convention did you use any odd symbols in your printer name?

VMaxx, 2019-04-04:
Ok I changed RepetierOutputDevice.py  Download it and see if it fixes your issue.

PPPBerlin, 2019-04-04:
> its requesting to see the job but no job is being returned. Do you have it set to print immediately or store?

I will only store the gcode. Printer name in repetier is CR-10S Pro. I download the file and now the plugin  doesn't appear in the printer menu of Cura 4.0

VMaxx, 2019-04-05:
typical location for the file is in: C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration

PPPBerlin, 2019-04-05:
I know and I i replaced the old file with your new file.

PPPBerlin, 2019-04-05:
Yes, but download the html site for the file is not the best choice ;) 
But after fixing that it still doesn't work. But i try to upload a file while printing. 

> 2019-04-05 09:19:34,739 - WARNING - [MainThread] UM.Logger.logException [88]:   File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\PrinterOutput\NetworkedPrinterOutputDevice.py", line 346, in __handleOnFinished
2019-04-05 09:19:34,743 - WARNING - [MainThread] UM.Logger.logException [88]:   File "C:\Program Files\Ultimaker Cura 4.0\plugins\RepetierIntegration\RepetierOutputDevice.py", line 661, in _onRequestFinished
2019-04-05 09:19:34,747 - WARNING - [MainThread] UM.Logger.logException [88]:     if "job" in json_data[0]:
2019-04-05 09:19:34,750 - WARNING - [MainThread] UM.Logger.logException [88]: KeyError: 0
2019-04-05 09:19:36,791 - WARNING - [MainThread] UM.Logger.logException [84]: Exception: something went wrong with callback
2019-04-05 09:19:36,795 - WARNING - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):

VMaxx, 2019-04-07:
if you downloaded it from the marketplace it goes here: C:\Users\{USER}\AppData\Roaming\cura\4.0\plugins\RepetierIntegration\RepetierIntegration

VMaxx, 2019-04-07:
I put an exception for when you try to print to a printer with a bad config and it times out, it will log an error saying printer misconfigured.

VMaxx, 2019-06-08:
I've updated a new version give it a try.

closed 2019-06-24 by VMaxx