VMaxx/RepetierIntegration #26: Adding new print to a different printer causes pauses on an already printing printer.

llambkin, 2019-04-17:
Hello VMaxx,

I have a new issue, I don't know if it is caused by the plugin or not. But if I am running a print on one machine, and I click print to repetier on a different printer, during the upload process, the printer that is already printing pauses for a second or 2, causing blobs on the print that is going.

Not a huge issue, but I was troubleshooting my little printer whilst the big one was printing a large job, and it had multiple blobs where I clicked print on the little printer.

Is there something in the process that would peg the processor of the Pi so much that it freezes the existing job while it catches up?

VMaxx, 2019-06-07:
Yeah I've found that since I only had one printer I coded for only one.  I'm working on fixing this currently.

VMaxx, 2019-06-08:
I just uploaded a new version that works with multiple printers.

closed 2019-06-24 by VMaxx