VMaxx/RepetierIntegration #28: Plugin crashes when trying to connect to printer

Jivix-zz, 2019-05-09:
Hi, just trying out the plugin for the first time. I was able to set up the printer, but connecting to it instantly crashes. I have tried using both a hostname and an IP address, and both port 80 and 3344 (the default repetier port), with no luck. I am running a fresh install of Cura 4.0.

I did not see anywhere if there is a required version of Repetier-Server, I am running 0.90.1, could that be the issue?

VMaxx, 2019-05-13:
Most of the issues people have is that when you create your printer in cura the name you use needs to match the one you give it in repetier.  I'm working on changing that but it will be some time before it's done.

Jivix-zz, 2019-05-14:
Hmm, maybe I should clarify. I understand the crashing if you change the information, but I get crashes even if I uninstall and reinstall everything, set up the printer, and try to connect to it.

Jivix-zz, 2019-05-14:
Hmm, I'm honestly not sure what the name of the printer needs to be. Which particular name in Repetier is the printer name that needs to be mirrored in Cura? Is it the "Printer Name" field in printer settings? If so, then it straight up doesn't work for me. Could it be because I'm using a hostname for my print server and not an IP address?

VMaxx, 2019-05-14:
Post logs so I can see

VMaxx, 2019-06-08:
I just uploaded a new version to try.

closed 2019-06-24 by VMaxx