VMaxx/RepetierIntegration #30: Checking the API key

YorickJanse, 2019-06-08:
When entering the API key, Plugin shows: Checking the API key ....

But nothing happens.

Cura 4.1

VMaxx, 2019-06-08:
Do you have security setup on repetier?

YorickJanse, 2019-06-09:
No, no security setup. When entering the API key the log shows:

2019-06-09 08:49:51,531 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [348]: http://192.168.2.125:80/printer/info
2019-06-09 08:49:51,533 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [349]: {'servername': 'Victoria-67', 'printers': [{'name': 'Anet A8 Plus', 'online': 1, 'slug': 'Anet_A8_Plus', 'active': True}], 'version': '0.91.2', 'apikey': '5053b99e-7f2c-4d99-95cb-b4547b2d166d', 'serveruuid': 'e0d86674-eec9-4737-9dea-735a31ed2fa3', 'name': 'Repetier-Server Free'}
2019-06-09 08:49:51,534 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [355]: DiscoverRepetierAction: printers: 1
2019-06-09 08:49:51,535 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [359]: Slug: Anet_A8_Plus

VMaxx, 2019-06-09:
looks like it is working, did you slice anything?

YorickJanse, 2019-06-10:
Nope, can not connect.  Connect button is grayed out, en text is: Checking the API key ...

VMaxx, 2019-06-11:
Only thing that seems curious from your log is port 80.  The default is 3344 unless you changed it on install.

VMaxx, 2019-06-12:
It should go something like this:
[image]
[image]
[image]

VMaxx, 2019-06-14:
It may be you need to hit enter on the API key entry.  Seems that triggers it.

YorickJanse, 2019-06-15:
Thanks for the suggestions, but I have no luck yet :-(

[image]

Moonlight63, 2019-06-16:
I came here to say I have the same problem. Can confirm this is an issue.

VMaxx, 2019-06-17:
check via your browser the http://192.168.2.125:3344/ make sure its right

Moonlight63, 2019-06-17:
In my case it is 10.0.1.110:3344. Also worth noting that cura makes it far enough to find the printers listed in Repetier Server when clicking on "Get printers". Side note: The instance name box doesn't allow for a name the same length as you may have for a printer. For example, I have a profile for my Prusa MK3, whose default name when adding to cura is "Original Prusa i3 MK3/MK3S" but I can only enter a name up to 20 characters long giving me "Original Prusa i3 MK" Not a big deal, but still. 
I've just checked on cura 4.0 and the problem persists. Is it possibly a naming issue? The name in repetier is "Prusa MK3" but the name retrieved with the get printers button is Prusa_MK3. Is the underscore an issue?

VMaxx, 2019-06-17:
I recently added an update to fix the drop down issue.   I also added a document that gives a step by step to add a printer in repetier and cura here: https://github.com/VMaxx/RepetierIntegration/blob/master/Cura%204.1%20testing%20doc.rtf.zip

VMaxx, 2019-06-17:
The underscore is correct.  Its what repetier refers to as the "slug" that it uses for the web calls.

YorickJanse, 2019-06-17:
Also notice that the repetierID in the picture is empty. In the example it shows Marvin1

VMaxx, 2019-06-18:
Its highlighted in blue indicating it is selected, followed by the next picture showing it as selected.

VMaxx, 2019-06-18:
did this help at all?

VMaxx, 2019-06-24:
I added a video to the read me
https://youtu.be/VHw93Pt_QIo

VMaxx, 2019-06-30:
Updated the getprinters button.  This was probably a big issue I never ran into since I hit enter instead of the OK button.  It should pick it up now.

maciekish, 2019-07-11:
I have the same problem, i can browse to the url, security is set up and correct details entered in plugin. Can get printer list but get stuck on "Checking the API key..."

YorickJanse, 2019-07-11:
Hi, 

With some debuging I have seen the following:

DiscoverRepetierActions.py:
- getPrinterList,  sometimes gets wrong base_url (http://<ip><port>/  instead of  <ip><port>
- testApiKey, work_id is empty

If I put a workaround in the code (work_id = <printer_name>, then I can connect.

But no:  Print with Repetier button :-(

I have a different setup: In the example video a windows server is used, I use a Raspberry Pi with Repetier Server

Ianmcmill, 2019-07-13:
I have the same issue. My url/ip/api is entered correct.

VMaxx, 2019-07-17:
I stumbled upon something that will pull the key if you have security off.

Ianmcmill, 2019-07-17:
Can I enable security again afterwards?

VMaxx, 2019-07-17:
Something to check, make sure RepetierID is showing something before you put the API key in.  If its empty then something didn't save your Printer selection under edit.

VMaxx, 2019-07-18:
I posted an update that should pull the apikey if you don't have security turned on.

VMaxx, 2019-07-18:
Also my main printer is on a raspberry pi so I've tested it on both windows and pi.

VMaxx, 2019-07-18:
I think I found the issue.  Make sure you click the get printers button so that it queries the instance and populates the dropdown.  If that hasn't been done it will perpetually give that message.  I'll adjust it shortly to fix.

VMaxx, 2019-07-19:
Ok I changed it so that you must actually click a valid entry in the combo box after you get printers on the add manual dialog.  This way it stores the repetierid properly.

YorickJanse, 2019-07-19:
Hi,

RepetierId is not showing:

[image]

Log file:

-19 12:22:21,783 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://192.168.2.125:3344/printer/info
2019-07-19 12:22:21,802 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [344]: http://192.168.2.125:3344/printer/info
2019-07-19 12:22:21,802 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [349]: http://192.168.2.125:3344/printer/info
2019-07-19 12:22:21,802 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [350]: {'name': 'Repetier-Server Free', 'serveruuid': 'e0d86674-eec9-4737-9dea-735a31ed2fa3', 'version': '0.91.2', 'servername': 'Victoria-67', 'printers': [{'name': 'Marvin', 'active': True, 'slug': 'Marvin', 'online': 1}, {'name': 'AnetA8Plus', 'active': True, 'slug': 'AnetA8Plus1', 'online': 1}], 'apikey': '5053b99e-7f2c-4d99-95cb-b4547b2d166d'}
2019-07-19 12:22:21,803 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [356]: DiscoverRepetierAction: printers: 2
2019-07-19 12:22:21,803 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [360]: Slug: Marvin
2019-07-19 12:22:21,804 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [360]: Slug: AnetA8Plus1
2019-07-19 12:22:21,804 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [364]: DiscoverRepetierAction: apikey: 5053b99e-7f2c-4d99-95cb-b4547b2d166d
2019-07-19 12:22:24,591 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://192.168.2.125:3344/printer/info
2019-07-19 12:22:24,611 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [344]: http://192.168.2.125:3344/printer/info
2019-07-19 12:22:24,611 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [349]: http://192.168.2.125:3344/printer/info
2019-07-19 12:22:24,612 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [350]: {'name': 'Repetier-Server Free', 'serveruuid': 'e0d86674-eec9-4737-9dea-735a31ed2fa3', 'version': '0.91.2', 'servername': 'Victoria-67', 'printers': [{'name': 'Marvin', 'active': True, 'slug': 'Marvin', 'online': 1}, {'name': 'AnetA8Plus', 'active': True, 'slug': 'AnetA8Plus1', 'online': 1}], 'apikey': '5053b99e-7f2c-4d99-95cb-b4547b2d166d'}
2019-07-19 12:22:24,612 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [356]: DiscoverRepetierAction: printers: 2
2019-07-19 12:22:24,613 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [360]: Slug: Marvin
2019-07-19 12:22:24,613 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [360]: Slug: AnetA8Plus1
2019-07-19 12:22:24,613 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [364]: DiscoverRepetierAction: apikey: 5053b99e-7f2c-4d99-95cb-b4547b2d166d
2019-07-19 12:22:28,868 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [97]: Repetier_ID: AnetA8Plus1
2019-07-19 12:22:28,868 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [124]: NAME IS: AnetA8Plus
2019-07-19 12:22:28,869 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [125]: ADDRESS IS: 192.168.2.125
2019-07-19 12:22:28,881 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.
2019-07-19 12:22:28,893 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.
2019-07-19 12:22:28,894 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://http//192.168.2.125:3344//printer/info
2019-07-19 12:22:29,726 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [340]: Received no or empty reply
2019-07-19 12:22:30,387 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.
2019-07-19 12:22:30,398 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.
2019-07-19 12:22:30,399 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://http//192.168.2.125:3344//printer/info
2019-07-19 12:22:31,224 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [340]: Received no or empty reply
2019-07-19 12:22:32,740 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://http//192.168.2.125:3344//printer/info
2019-07-19 12:22:33,566 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [340]: Received no or empty reply
2019-07-19 12:22:38,875 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles

Notice the: getPrinterList:http://**http//**192.168.2.125:3344**/**/printer/info

VMaxx, 2019-07-19:
yeah it's odd.  I'll look into that see why its doing that for you and not me.

VMaxx, 2019-07-19:
its curious because at the top of the log, it is using the correct url, but at the bottom it isn't

VMaxx, 2019-07-19:
Its like the cura config for the printer isn't there, so it can't save the information.
this message here makes me think something is out of alignment:
2019-07-19 12:22:28,881 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.

This would also prevent you from changing other settings in your printer, if you've been having problems with that.
Check your %AppData%\cura\4.1\machine_instances folder.

If its not too difficult I'd just uninstall and reinstall cura and set the printer back up.

YorickJanse, 2019-07-20:
Reinstalled Cura,same problem :-(

VMaxx, 2019-07-21:
Check your %AppData%\cura\4.1\machine_instances folder they're text files. open them in notepad see if you have a repetier_ anything in there.

YorickJanse, 2019-07-21:
[general]
version = 4
name = AnetA8Plus
id = Custom FFF printer

[metadata]
repetier_id = AnetA8Plus1
group_id = 477af5c2-84da-49b0-83ae-fbe508c3cf7c
type = machine
repetier_api_key = 5053b99e-7f2c-4d99-95cb-b4547b2d166d
setting_version = 7

[containers]
0 = Custom FFF printer_user
1 = empty_quality_changes
2 = normal
3 = empty_material
4 = empty_variant
5 = Custom FFF printer_settings
6 = custom

VMaxx, 2019-07-21:
Its got the info correct.  So it some other issue.  I'll create a separate version with your name on it, for testing shortly.

VMaxx, 2019-07-21:
Ok I just posted the change.  It may have been some unneeded code doing something.  So I commented out a small section.

VMaxx, 2019-07-22:
It's here:
https://github.com/VMaxx/RepetierIntegration/tree/YorickJanse

YorickJanse, 2019-07-23:
2019-07-23 19:46:41,405 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://192.168.2.125:3344/printer/info
2019-07-23 19:46:41,427 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [345]: http://192.168.2.125:3344/printer/info
2019-07-23 19:46:41,427 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [350]: http://192.168.2.125:3344/printer/info
2019-07-23 19:46:41,427 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [351]: {'apikey': '5053b99e-7f2c-4d99-95cb-b4547b2d166d', 'name': 'Repetier-Server Free', 'servername': 'Victoria-67', 'serveruuid': 'e0d86674-eec9-4737-9dea-735a31ed2fa3', 'version': '0.91.2', 'printers': [{'name': 'Marvin', 'active': True, 'slug': 'Marvin', 'online': 1}, {'name': 'AnetA8Plus', 'active': True, 'slug': 'AnetA8Plus1', 'online': 0}]}
2019-07-23 19:46:41,428 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [357]: DiscoverRepetierAction: printers: 2
2019-07-23 19:46:41,428 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [361]: Slug: Marvin
2019-07-23 19:46:41,428 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [361]: Slug: AnetA8Plus1
2019-07-23 19:46:41,429 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [365]: DiscoverRepetierAction: apikey: 5053b99e-7f2c-4d99-95cb-b4547b2d166d
2019-07-23 19:46:44,862 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://192.168.2.125:3344/printer/info
2019-07-23 19:46:44,887 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [345]: http://192.168.2.125:3344/printer/info
2019-07-23 19:46:44,888 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [350]: http://192.168.2.125:3344/printer/info
2019-07-23 19:46:44,888 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [351]: {'apikey': '5053b99e-7f2c-4d99-95cb-b4547b2d166d', 'name': 'Repetier-Server Free', 'servername': 'Victoria-67', 'serveruuid': 'e0d86674-eec9-4737-9dea-735a31ed2fa3', 'version': '0.91.2', 'printers': [{'name': 'Marvin', 'active': True, 'slug': 'Marvin', 'online': 1}, {'name': 'AnetA8Plus', 'active': True, 'slug': 'AnetA8Plus1', 'online': 0}]}
2019-07-23 19:46:44,889 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [357]: DiscoverRepetierAction: printers: 2
2019-07-23 19:46:44,889 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [361]: Slug: Marvin
2019-07-23 19:46:44,889 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [361]: Slug: AnetA8Plus1
2019-07-23 19:46:44,890 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.DiscoverRepetierAction._onRequestFinished [365]: DiscoverRepetierAction: apikey: 5053b99e-7f2c-4d99-95cb-b4547b2d166d
2019-07-23 19:46:48,960 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.RepetierOutputDevice.__init__ [97]: Repetier_ID: AnetA8Plus1
2019-07-23 19:46:48,961 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.RepetierOutputDevice.__init__ [123]: NAME IS: AnetA8Plus
2019-07-23 19:46:48,961 - DEBUG - [MainThread] RepetierIntegration-YorickJanse.RepetierOutputDevice.__init__ [124]: ADDRESS IS: 192.168.2.125
2019-07-23 19:46:48,974 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.
2019-07-23 19:46:48,982 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.
2019-07-23 19:46:50,573 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.
2019-07-23 19:46:50,582 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [67]: Could not get metadata of container AnetA8Plus because it was not found.
2019-07-23 19:46:58,965 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles

VMaxx, 2019-07-23:
This a windows box?

VMaxx, 2019-07-24:
do you have any of these? RepetierIntegration.RepetierOutputDevice.connect

YorickJanse, 2019-07-25:
Found the problem!
If you install Cura, and add a printer. The default name will be Custom+fff+printer.global.cfg
(The name input screen while doing setup is hard to see) 
When renaming the printer after setup, the name in the .cfg file is changed, the id is still 'Custom FFF printer'
Somehow this does not work for the RepetierIntegration.

To fix the problem: Delete the Custom+fff+printer.cfg
Start Cura. Setup will start again, and make sure you enter a printer name in the setup screen.

VMaxx, 2019-07-25:
awesome!

VMaxx, 2019-07-25:
I suspect this might be a cura issue in general as research on it shows people having this same issue before I started.  It may be worth escalating to ultimaker.

YorickJanse, 2019-07-26:
Thanks for the support, it is realy nice to print directly from Cura without saving the file first :-)

VMaxx, 2019-07-27:
Sure thing, are you still using the version I named for you, or the regular?

YorickJanse, 2019-07-27:
I'm using the version specially made for me

davidroland78, 2019-11-12:
hi.to cura 4.3 is the same problem checking the API key ... and is not finishing.i changed thr 80 with 3344 and nothing? to delete that file which you suggested for cura 4.1? where to find it?

davidroland78, 2019-11-14:
problem SOLVED .