VMaxx/RepetierIntegration #32: Cant connect to some printers

userosos, 2019-08-12:
I can not connect to the some printers because repetier ID cannot apply.
[image]
[image]

VMaxx, 2019-08-13:
Check the cura.cfg file in %Appdata%\cura see if its recording a Repetier_id
It should look something like this:
`manual_instances = {"Tarantula": {"userName": "", "path": "/", "address": "192.168.1.250", "useHttps": false, "repetier_id": "Tarantula", "password": "", "port": 3344}, "Fred": {"userName": "vmaxx", "path": "/", "address": "192.168.1.250", "useHttps": false, "repetier_id": "Fred1", "password": "FooFoo", "port": 3344}}`

timoschuett, 2019-08-13:
Same problem here. I got a repetier_id in the cfg but it shows the same error like @userosos

VMaxx, 2019-08-15:
Ok, I'll have to throw in some extra debugging to figure out what's going on.

trelowney, 2019-08-16:
I have same problem with it. Last working version is 4.0.0 for me, newer versions show this problem with drop down list. Cfg file is ok. When I copy repetier part from working 4.0.0 to 4.2.1 or 4.1, it's same and no connect. Many restart of Cura and testing and don't work. Repetier-Server Free 0.91.2 on ARM..  :-/

(Video of problem: https://www.youtube.com/watch?v=fVmv-rrpMbM&feature=youtu.be )

VMaxx, 2019-08-16:
Thanks, the video was helpful.  For grins, try hitting the enter key instead of Ok and see if it behaves differently.

VMaxx, 2019-08-17:
also try clicking the tronxy in the list before putting in the API, mine shows blank as soon as I add the printer and doesn't reload until I click from the list to type in the API.

userosos, 2019-08-17:
@VMaxx 
In cura.cfg i can see
[Repetier]
manual_instances = {"Ulti": {"password": "", "userName": "", "address": "192.168.0.23", "useHttps": false, "repetier_id": "Ulti", "path": "/", "port": 3344}, "Sprinter": {"password": "", "userName": "", "address": "192.168.89.2", "useHttps": false, "repetier_id": "Sprinter", "path": "/", "port": 80}}
keys_cache =

trelowney, 2019-08-17:
> also try clicking the tronxy in the list before putting in the API, mine shows blank as soon as I add the printer and doesn't reload until I click from the list to type in the API.

It's same with enter key. If you want, see the video. (in Cura 4.0.0 is another username, but i test it also with same username.) 
I can test anything further :)

YT: https://www.youtube.com/watch?v=hTSw3KUFMi4

VMaxx, 2019-08-19:
@userosos  port 80 on Sprinter should be 3344.
@trelowney I've not been able to replicate this from testing.  Can you e-mail your %AppData%\cura\4.2\cura.log file to vmaxx@cableone.net  Thanks!

userosos, 2019-08-19:
> @userosos port 80 on Sprinter should be 3344.

I set the port to 80 but 3344 not work too.

VMaxx, 2019-08-20:
Ok I made a change to the instance selection part that will generate an extra log if it returns nothing.  Try it out and then send me your log file again.  Thanks!

trelowney, 2019-08-20:
> Ok I made a change to the instance selection part that will generate an extra log if it returns nothing. Try it out and then send me your log file again. Thanks!

Sended :)

userosos, 2019-08-20:
I add the log also for more information.
https://www.sendspace.com/file/sm4dtf

VMaxx, 2019-08-21:
Check if your %AppData%\Roaming\cura\4.2\machine_instance file is named "Custom FFF printer.global.cfg" and not Tronxy.global.cfg

trelowney, 2019-08-22:
> Check if your %AppData%\Roaming\cura\4.2\machine_instance file is named "Custom FFF printer.global.cfg" and not Tronxy.global.cfg

It is named Custom+FFF+printer.global.cfg

VMaxx, 2019-08-22:
I have verified, if you create a custom printer and leave the default name, then rename it.  It causes the call cura.Settings.ContainerManager.getContainerMetaDataEntry to fail.

VMaxx, 2019-08-23:
Try renaming your Custom+FFF+printer.global.cfg to tronxy.global.cfg, I think it will resolve the problem.

trelowney, 2019-08-23:
Yes, this solve the problem. Very thanks for this ;)

userosos, 2019-08-23:
I fix the problem too. Now cura crash if i change an printer in monitor window. How i can fix the problem?

VMaxx, 2019-08-24:
@trelowney awesome.  I'm going to post this to Cura.  They've got a "feature" when renaming printers.
@userosos delete your current cura.log file then run this program and send me the new smaller cura.log at vmaxx@cableone.net

userosos, 2019-08-24:
@VMaxx I can not repeat the bug now. It work OK now.

VMaxx, 2019-08-26:
Awesome!

timoschuett, 2019-08-30:
fixed the selection problem too. now cura crashes when i enter the api key.