VMaxx/RepetierIntegration #33: Repetier printer name hasn't been selected

betavr, 2019-08-20:
I'm trying to get the plugin to work under Cura 4.2.1, but no cake.

After pasting the API key in the "Connect Repetier" window, this message is displayed under the input field:

> Repetier printer name hasn't been selected, please edit and click printers and choose the correct name from the drop down list.

I followed the instructions, clicked the Edit button, then clicked the "Get Printers" button in the Edit dialog. The list is populated correctly with the printer names configured in my Repetier-Server instance. However after selecting the printer and pressing OK, the API key field still gives me the same error message. Opening the Edit dialog again reveals that the Printer dropdown is again empty, so it didn't save my selected printer.

VMaxx, 2019-08-22:
Check if your %AppData%\Roaming\cura\4.2\machine_instance file actually matches your machine instance name.  Common issue is taking a default, then renaming it causing issues.

smolz, 2019-10-25:
Seeing the same thing, but noticed this in the logs:

This specifically: getPrinterList:http://http//10.0.40.210:3344//printer/info

2019-10-25 14:28:48,195 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://http//10.0.40.210:3344//printer/info
2019-10-25 14:28:48,200 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [340]: Received no or empty reply
2019-10-25 14:28:48,701 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:**http://http//10.0.40.210:3344//printer/info**
2019-10-25 14:28:48,703 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [340]: Received no or empty reply

vair327, 2019-10-27:
Same issue here.

vair327, 2019-10-27:
Same issue here with 4.3.

XKiloX95, 2019-10-27:
Same issue here with no luck in 4.3

peeksy, 2019-11-03:
same for me. It seems the server doesn't send the printer name. However, when I press "Get Printer" a few times a name appears bu this is not the name of the printer as it is defined in RepetierServer

dgehri, 2019-11-11:
Is there any update to this? I'm experiencing the same issue. Initially, I thought it was due to a space character in the printer name, but I since recreated the same printer with a simple name and this doesn't fix the problem.

VMaxx, 2019-11-11:
In an effort to fix one problem another happened.  Try re-downloading it.

dgehri, 2019-11-12:
> In an effort to fix one problem another happened. Try re-downloading it.

I'm afraid, but the issue is present in the last commit (4b3b57e0dcd83decce56b76dfd9b72a63048db72).

downtown75, 2019-11-12:
Same here with Cura 4.3. Printer show up in the drop-down list but cannot assign API Key (printer name hasn't been selected).

mcpea1477, 2019-11-16:
I tried cura 4.3 and 4.1 on mac...same Problem on both...

VMaxx, 2019-11-17:
Check %appdata%\Roaming\cura\4.3\machine_instances and see if your cfg file matches your printer name.  This is often the problem with this particular error.  This often occurs when you start with one name, then use the rename button in cura.  It has a bug where it renames the content, but doesn't rename the file, so when plugins go to access it, it fails.

downtown75, 2019-11-17:
True, but I'm kind of fordced to rename as the amount of characters in "Instance Name" is limited (to 20 (?). But the name for my local printer is 23 characters long.

What worked for me:
. Delete this Plugin files from Cura Programm and/or Plugin directory (%appdata%).
. Delete the machine instance(s) from directory %appdata%\Roaming\cura\4.3\machine_instances
. Start Cura and the Assistant to add a printer should popup.
  Important: Select and **Rename the Printer name here** to maximum 20 characters!
  (In my case the printer name was "Artillery Sidewinder X1" and I renamed it to "Sidewinder X1"
. Follow Plugin Howto; printer should appear now and be selectable

Thanks for your help VMaxx!

mcpea1477, 2019-11-18:
> Check %appdata%\Roaming\cura\4.3\machine_instances and see if your cfg file matches your printer name. This is often the problem with this particular error. This often occurs when you start with one [...]

Hello VMaxx,

Thank you...it worked for 4.1...no time yet to check 4.3 but I guess will work too.
My biggest problem was to find this cfg in mac OS. Means the equivalent of %appdata%. Found it, changed the name, worked fine! Thanks!

Tobilott, 2020-03-09:
Hello VMaxx,

i got the same problem with Cura 4.5. I checked the %appdata%\Roaming\cura\4.5\machine_instances. And the name was correct. 
Printer show up in the drop-down list but cannot assign API Key (printer name hasn't been selected).

misterbike, 2020-03-13:
Hello VMaxx,

i've got the same problem like Tobilott....Plugin dosen't work.
I tried with cura 4.4.1 and 4.5 .... :-(

Tobilott, 2020-03-13:
Hi it was my fault. I used the master version instead of the version 4.5. After installing the 4.5 it works fine.

misterbike, 2020-03-13:
Jippieh ..... it works :-)
Thanks very much for this important information.
I've looked the video and there was also the "master" Version shown...so i thougt, it was OK

VMaxx, 2020-03-17:
Good to know.  The master version has changes I made when I attempted to adjust the dialog size for the Macs.

kolossboss, 2020-03-19:
Hi, 
Unfortunately I'm heaving the same error.

I'm using Cura 4.5 and latest branch of the cura plugin (v4.5)

There are two printers connected to the repetier server. Both have this problem.

I think I tried everything in this thread without any luck.

In the Cura Marktplace, the plugin shows v4.4 but in the readme its v4.5

Thats the cura "machine instances" config file

`[general]
version = 4
name = Creality Ender-3
id = Creality Ender-3

[metadata]
connection_type = 2
mks_network_key = manual:192.168.2.156
octoprint_show_camera = true
repetier_api_key = 2c9ca241-b277-4bb5-af94-xxxxxxxxx
octoprint_store_sd = false
group_id = 5ac7456f-1004-4710-b712-xxxxxxx
type = machine
setting_version = 11
octoprint_api_key = MzQ2REMxOUUyNUQ4NDg2Oxxxxxxxxxxx
post_processing_scripts = 
octoprint_id = 
repetier_id = Ender_3

[containers]
0 = Creality Ender-3_user
1 = creality_ender3_ender_3_für_windows_pc #2
2 = empty_intent
3 = base_global_draft
4 = empty_material
5 = empty_variant
6 = Creality Ender-3_settings
7 = creality_ender3
`
[image]

Thanks for helping

psypher246, 2020-03-30:
> Hi,
> Unfortunately I'm heaving the same error.
>
> I'm using Cura 4.5 and latest branch of the cura plugin (v4.5)
>
> There are two printers connected to the repetier server. Both have this problem.
>
> I [...]

I had the same problem until I named everything Da_Vinci becuase that's how it was being listed when I clicked get printers. 

This is what I did:
Remove the printer and make sure the machine instances config file is also removed.
Create new printer, name it Ender_3 (note the underscore because that's how it showed up in your list of printers)
Add a new instance of the printer, also named Ender_3, select the printer from the list
Add API key and boom it worked for me.

Incidentally, no matter what I named the printer on Repertier server, it always listed as Da_Vinci. I tried naming it DaVinci, XYZ, Da Vinci, it always listed as Da_Vinci so I assume it's pulling that from Repetier Host and not Server.

However I have a new problem now where I send the print to Repetier, it says the printer is busy (it's not) and then asks to queue it, it "uploads" but nothing happens, nothing in the queue of repetier server either.
Cura 4.5
Repertier Server 0.93.1
Da Vinci XYZ 1.0

kolossboss, 2020-03-31:
@psypher246 
thanks for reply. Thats the only thing I did not, because I'm afraid of loose all of my profiles and settings.
I'm now using the Repetier Server Monitor the directly upload the gcode to the Repetier Server und I started the print via the my Touchscreen. This works great and it also works with all slicers.

Krisshellman1, 2020-04-24:
Same issue with 4,5 and 4.6....

FindusD, 2020-04-25:
I have the same problem.
When I click on Get Printers I get only the name of the printers mainboard GT2560- not the defined name of the printer.
But even when I rename my printer to GT2560 I cannot connect, all options are grey

kolossboss, 2020-10-12:
Still having this error with my Ender 3.
Today I added a second Printer to repetier server and tried to connect to the Cura Plugin. AND its working perfect and instantly. Why is it not working with my Ender 3? Whats the differents.

VMaxx, 2020-10-29:
Difference is your ender 3 was likely renamed causing the issue in cura.  The only fix I have is to open the settings for it, write it all down.  Delete the printer and re-add it without renaming.

llambkin, 2023-05-10:
I find it so odd, that this plugin has been updated so many times to keep up with Cura, and there has still been no fix to this issue. Surely there has to be an easier way than deleting the printer, and making sure the text in the name on Cura and in the repetier server match exactly.....

VMaxx, 2023-05-22:
The problem seems to be how cura tracks the names of printers.  If it
doesn't match with repetier and you rename it, internally cura keeps the
original name causing problems.

aedificatar, 2023-12-08:
I have a clean install of Cura 5.6.0 installed, installed the Repetier plugin per instructions, the plugin is able to query the server (running on a RPI4B) for printer list successfully, I select it, put in current API key - even changed it couple times to test and Immediately get this message: "Repetier printer name hasn't been selected, please edit and click get printers and choose the correct name from the drop down list".

Are there any suggestions on how to fix this @VMaxx or anyone else?  Thanks.

VMaxx, 2023-12-09:
The most common issue is creating the name in repetier different than what
you name it in Cura.  Cura has a means to rename, but it doesn't work
properly.  You have to delete the printer and re-add it with the matching
repetierid name.

aedificatar, 2023-12-14:
Thank you for the reply.  I got the same problem in Cura after completly uninstalling & reinstalling Cura v5.6.0, deleted & recreated - different name from before the printers on Repetier Server.  Is there something else I can try by chance?  Can it be Windows firewall related by chance?  I can use Repeiter Host, but its not nearly as clean and easy to use as Cura!

Thanks.

VMaxx, 2023-12-14:
The name you gave it in repetier server must match what you use in cura.

aedificatar, 2023-12-15:
The name is the exact same in both characters and case with no name change from when created on Repetier to Cura where it gave me the error.  Got a strong feeling this isn't related to difference in the name.  Any other ideas?

VMaxx, 2023-12-20:
email your log file to me at ***@***.***

ayaromenok, 2024-02-11:
I have exactly the same issue with Cura 5.6, RepetierIntegration master and Repetier Server 1.4.15 (also running on Raspberry Pi, like @aedificatar)

Plz find my cura.log below 
[cura.log](https://github.com/VMaxx/RepetierIntegration/files/14233907/cura.log)

ayaromenok, 2024-02-12:
Maybe related to bug issue - 1 position in printer list is empty
[image]

VMaxx, 2024-02-14:
I ran this on Ubuntu this evening with the same name etc.  You should just have to click on Tronxy_i3 from the drop down there then hit ok.  However, if you setup a password on your repetier, you have to click show security options and put your login to the repetier server there.  That may be the issue.

EmEiBee, 2024-04-18:
... there are some more where it can go wrong (windows 10).
1st the old folder from former CURA versions with the same config-files...
2nd the wrong names for the printer in \Users\[user]\AppData\Roaming\cura\[version]\machine_instances\[printername].global.config

My way:
1. (ZIP) and delete folder of old cura versions behind \%appdata%\cura
2. Open \%appdata%\cura\[version]\machine_instances\[printername].global.config
3. Fix inside the wrong names and, if not exist, place the API key from RS in

I have set **_name_** and **_id_** the same, that have to match _**repetier_id**_

I also renamed the cfg itself but have to start CURA twice (1st start CURA try to start the "1st printer entry thing")

The only thing I don't know is in section [containers] what 0=[some printername]_user means so I haven't touch that wrong printer name there...