VMaxx/RepetierIntegration #36: Cura 4.3 for mac layout

smolz, 2019-10-25:
In Cura 4.3 for Mac, you can only barely make out the buttons at the bottom.
[image]

VMaxx, 2019-10-29:
That's not a cool feature is it?  I'll see what I can do to fix it.

VMaxx, 2019-11-02:
I uploaded a new DiscoverRepetierAction.qml see if that helps.

VMaxx, 2019-11-05:
Testing this in elementary linux yielded bad results so I'm going to roll the change back.

VMaxx, 2019-11-05:
the work around for now is to grab the window and resize it to see the buttons.  The close button should float but the others are static.  The qml is supposed to autosize to fit, but its not working properly here in OSx for whatever reason.