VMaxx/RepetierIntegration #37: add support cura 4.4 please

userosos, 2019-11-19:
Add support cura 4.4, please. It hase some changes for addons.

VMaxx, 2019-11-21:
I added a 4.4 branch its not in master yet.  Let me know how it goes.

downtown75, 2019-11-22:
Works for me with Cura 4.4.

CageFox, 2019-12-10:
I was unable to use it in 4.4. When I creating new connection there is infinite loop:

1. I choosed printer name and press OK
2. Enter API key, it tell me "no printer name has been selected"
3. I press Edit, there is printer selected by Ok is inactive. I select printer, press OK, field for api key become empty and step 2 again

Beliar-666, 2019-12-10:
Me unable to connect. Same as Cagefox.

downtown75, 2019-12-10:
see this link; this i how it worked for me:
https://github.com/VMaxx/RepetierIntegration/issues/33#issuecomment-554793830
have you tried this procedure?

CageFox, 2019-12-10:
Response for my http://pi:3344//printer/info/
{"apikey":"0769a27c-9eaf-4008-a912-a7b901408dd9","name":"Repetier-Server Free","printers":[{"active":true,"name":"4MP","online":0,"slug":"C"}],"servername":"C","serveruuid":"e0c95a00-db68-4f60-bd4d-56700d75f58f","version":"0.91.2"}

I expect I will see 4MP printer in list. But I see C in list

May be you take "slug" from response as printer name? I think it better to take "name"

CageFox, 2019-12-10:
I was able to connect when I specify 4MP as CONTAINER for printers. But this is not right, plugin must read container cura.Settings.ContainerManager.getContainerMetaDataEntry with printer name, not slug

CageFox, 2019-12-10:
And I changed code a little:
                            for printerinfo in json_data["printers"]:
                                 Logger.log("d", "Slug: %s",printerinfo["slug"])
                                 self._printers.append(printerinfo["**name**"])

CageFox, 2019-12-10:
No, this patch helps me just make thru configuration interface, plugin then use printer name instean of repetier instance name

Sangoku, 2019-12-22:
Same here. i insert API key. connect to server, search a printer, pick one and the API key is gone. 
So i cant click to connect. because the auto start ist disabled im guessing.

CageFox, 2020-03-03:
This problem is still not fixed. Is the plugin dead?

CageFox, 2020-03-03:
Ah, sorry, i didn't noticed 4.4 branch. With 4.4 I can see printer in monitor it's much better

closed 2020-04-28 by VMaxx