VMaxx/RepetierIntegration #39: Infinite loop to set up

Sangoku, 2019-12-28:
There is an infinite loop where i am stuck at. 

Tested on 4.3 and 4.4 same result... 

steps to reproduce:
click manage printers -> 
[image]
Cant connect to the printer. it says you have ot pick the printer.. which i do.. 
when i do pick it from the dropdown the API key vanishes... 
And you can start over.. 

Here is the error log from cura:

`2019-12-28 23:23:35,648 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\default.shader]...
2019-12-28 23:23:35,651 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\selection.shader]...
2019-12-28 23:23:35,654 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\select_face.shader]...
2019-12-28 23:23:35,657 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\default.shader]...
2019-12-28 23:23:35,660 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\composite.shader]...
2019-12-28 23:23:35,663 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\overhang.shader]...
2019-12-28 23:23:35,667 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\striped.shader]...
2019-12-28 23:23:35,671 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\transparent_object.shader]...
2019-12-28 23:23:35,674 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\striped.shader]...
2019-12-28 23:23:35,678 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\default.shader]...
2019-12-28 23:23:35,681 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\grid.shader]...
2019-12-28 23:23:35,684 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.3\resources\shaders\platform.shader]...
2019-12-28 23:23:35,757 - DEBUG - [MainThread] cura.Machines.QualityManager.initialize [136]: Lookup tables updated.
2019-12-28 23:23:35,856 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromIndexedVertices [485]: Calculating normals took 0.0019943714141845703 seconds
2019-12-28 23:23:35,868 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [61]: Updating QualityProfilesDropDownMenuModel.
2019-12-28 23:23:35,876 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-12-28 23:23:36,432 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [714]: MKS wifi reconnecting
2019-12-28 23:23:36,432 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:36,433 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.connect [154]: MKS socket connecting 
2019-12-28 23:23:36,433 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:36,911 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 2.5s
2019-12-28 23:23:40,427 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [714]: MKS wifi reconnecting
2019-12-28 23:23:40,427 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:40,428 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.connect [154]: MKS socket connecting 
2019-12-28 23:23:40,428 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:44,425 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [714]: MKS wifi reconnecting
2019-12-28 23:23:44,425 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:44,426 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.connect [154]: MKS socket connecting 
2019-12-28 23:23:44,426 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:45,752 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2019-12-28 23:23:46,103 - DEBUG - [MainThread] cura.Machines.QualityManager.initialize [136]: Lookup tables updated.
2019-12-28 23:23:46,133 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromIndexedVertices [485]: Calculating normals took 0.0019948482513427734 seconds
2019-12-28 23:23:46,206 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [61]: Updating QualityProfilesDropDownMenuModel.
2019-12-28 23:23:46,213 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-12-28 23:23:48,427 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [714]: MKS wifi reconnecting
2019-12-28 23:23:48,427 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:48,428 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.connect [154]: MKS socket connecting 
2019-12-28 23:23:48,428 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:50,577 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.3/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
2019-12-28 23:23:52,424 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [714]: MKS wifi reconnecting
2019-12-28 23:23:52,425 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:52,425 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.connect [154]: MKS socket connecting 
2019-12-28 23:23:52,426 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:56,424 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [714]: MKS wifi reconnecting
2019-12-28 23:23:56,424 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:56,425 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.connect [154]: MKS socket connecting 
2019-12-28 23:23:56,425 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:23:58,483 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Users/sinis/AppData/Roaming/cura/4.3/plugins/RepetierIntegration/DiscoverRepetierAction.qml:655: TypeError: Type error
2019-12-28 23:23:58,485 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.3/qml/UM/Dialog.qml:84:13: QML Loader: Binding loop detected for property "active"
2019-12-28 23:24:00,425 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [714]: MKS wifi reconnecting
2019-12-28 23:24:00,425 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:24:00,426 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.connect [154]: MKS socket connecting 
2019-12-28 23:24:00,426 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin._onPrinterConnectionStateChanged [153]: mks add output device False
2019-12-28 23:24:04,426 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [714]: MKS wifi reconnecting
[... cut ...]
2019-12-28 23:25:26,706 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [155]: getPrinterList:http://http//192.168.0.6:3344//printer/info
2019-12-28 23:25:27,012 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [340]: Received no or empty reply
2019-12-28 23:25:27,013 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [340]: Received no or empty reply
2019-12-28 23:25:27,013 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [340]: Received no or empty reply`

Schwuuuuup, 2020-02-28:
I have the same Problem on Linux (AppImage)
My Printer was Named "TronXY X1 an Minion" (with cases and spaces)
So I renamed it to something simpler in my Repetier-Server, But still the same problem AND after pressing "Get Printers" I still get the OLD name of the printer in the drop down menu.

I do not know if the Repetier Server sends the old name somehow (although it doesn't look like it)
After Deleting the Cura config dir I had to re-initialize Cura, but the old printer config was still in the Repetier Integration menu. Where is it stored?

VMaxx, 2020-03-02:
seems to be under your home /.local/share/cura/4.5/machine_instances

Schwuuuuup, 2020-03-02:
> seems to be under your home /.local/share/cura/4.5/machine_instances

No, the stubborn printer name was actually in the Repetier Service... Seem you can't thoroughly rename a printer. 

So I deleted all settings and created a new printer (losing all my stored models) but that didn't help, it still says I have no printer name selected when I try to store the api key

MoellerDi, 2020-03-16:
same here with current master (d3df3746552c349cd76f8e1f688fffb1e57790e1) and Cura 4.5 on macos

thjubeck, 2020-04-17:
If you are having this issue delete your %apdata%\Roaming\cura folder and when it starts over, changed the name of the printer before you start. Use this as the instance name when connecting or it will not work. NOTE: This erases all Cura settings!!!!!!!

Schwuuuuup, 2020-04-20:
> If you are having this issue delete your %apdata%\Roaming\cura folder and when it starts over, changed the name of the printer before you start. Use this as the instance name when connecting or it will not work. NOTE: This erases all Cura settings!!!!!!!

I deleted everything and reinstalled Cura 4.5 same behavior:

1. Install Cura
2. Start Cura (Folders in Roaming get created)
3. Setup Custom FFP Printer, because Cura needs a Printer config to begin with
4. Stop Cura
5. Extract RepetierIntegration into Plugin Derectory
6. Start Cura again
7. Manage Printers 
8. Connect Repetier
9. Enter Instance Name (Not sure if this has to me something specific ore just a name I can choose)
10. Enter IP (and username/password) of RepetierServer
11. click "get printers" (this seems to be a bit faulty, you cannot see that some thing ist happening, the Printer selection stays empty at first but when I am close to giving up, suddenly there is a printer to select)
12. Save
13. Now there is a field to enter the API-Key, but when I enter one, there is always the message that there wasn't a name given
14. Rinse and repeat

Hope this is detailed enough to spot the mistake I'm making

thjubeck, 2020-04-20:
When you start cura and add the printer the first time, the name that you give the printer  will be the instance name in #9.