VMaxx/RepetierIntegration #40: Cura 4.5 Install not working

Thomas-Ganter, 2020-02-28:
As per instructions, placed the `RepetierIntegration` folder inside the plugins folder `/Users/thomas/Library/Application Support/cura/4.5/plugins` but I see not Connect button in Manage Printers. 

Most likely because of the version mismatch. 

Also don’t find a branch labelled 4.5, so I suspect nothing is there yet. How long until 4.5 is supported?

VMaxx, 2020-03-02:
Easiest thing to do is right click the start/window lower left and click run.
Type in %Appdata%\cura and hit enter.
This will get you to the correct folder to find the plugins.

Blueforcer, 2020-04-02:
Same problem here.
Copied it to 

> AppData\Roaming\cura\4.5\plugins\RepetierIntegration\RepetierIntegration

Its also listed as installed plugin, but no connect button:(
[image]

thjubeck, 2020-04-17:
If you are having this issue delete your %apdata%\Roaming\cura folder and when it starts over, changed the name of the printer before you start. Use this as the instance name when connecting or it will not work. NOTE: This erases all Cura settings!!!!!!!

VMaxx, 2020-04-28:
A major problem I've found is that if you've ever renamed a printer, it doesn't rename the underlying files.  So when  you query via the plugin code it never gets the correct information.  Making sure you clear those files out and start so that the names match from the beginning fixes the problem.