VMaxx/RepetierIntegration #42: Cura 4.6.1 connection won't work

pforrmi, 2020-05-03:
Hi all,

with cura v 4.6.1 the connection won't work. Any suggestions?

Kind regards
Michael

VMaxx, 2020-05-06:
Click on your printer button, click manage printer
highlight your printer click connect repetier
Then click connect

VMaxx, 2020-05-07:
I'm working on another update.  Work has been difficult for a while so looking at the Octoprint plugin to compare hasn't been done in quite a while.  This plugin is basically a clone of the Octoprint plugin with repetier commands in place.

VMaxx, 2020-05-08:
Ok there's a 4.6.1 out there now I tested with my setup.  It seems to function better than it used to.

TobiasKarge, 2020-06-22:
- i install the plugin
- manage the printer 
- klick on "connect Repetier"
- klick "add"
- enter the Repetier IP
- klick "Get Printer"
- choose the printer in dropdown List
- klick "OK"
- enter the API Key and the plugin say "Repetier printer name Hasen't been selected, please edit..."
- so i edit and get printer and enter the API Key in an infinite loop

serco351, 2020-06-23:
@TobiasKarge Make sure the names are correct both should be the same
Curaprinter name: My Printer
Repetier Plugin: My Printer
[image]

TobiasKarge, 2020-06-24:
Thanks it works

bpivk, 2020-06-27:
Whyt If my printer is named Creality Ender 5 Plus and the name is too long? Can you modify it so that it can accept longer names?

baradhili, 2020-06-30:
I think that image needs to be in the readme - also I get the same issue as bpivk - printer name is too long

VlaoMao, 2020-08-04:
Doesn't working on Cura 4.6.2, tricks doesn't help, in stderr i see log:

2020-08-04 19:03:39,990 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://192.168.1.24:3344/printer/info
2020-08-04 19:03:39,994 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [469]: http://192.168.1.24:3344/printer/info
2020-08-04 19:03:39,994 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [474]: http://192.168.1.24:3344/printer/info
2020-08-04 19:03:39,995 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [475]: {'apikey': '*****-****-****-****-3c9df91501b4', 'version': '0.93.1', 'servername': 'Jupiter-60', 'name': 'Repetier-Server Free', 'printers': [{'active': True, 'online': 1, 'slug': 'tevo', 'name': 'tevo'}], 'serveruuid': '*****-****-****-****-3c9df91501b4'}
2020-08-04 19:03:39,995 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [481]: DiscoverRepetierAction: printers: 1
2020-08-04 19:03:39,996 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: Slug: tevo
2020-08-04 19:03:39,996 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [489]: DiscoverRepetierAction: apikey: *****-****-****-****-3c9df91501b4
2020-08-04 19:03:41,588 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [100]: Repetier_ID: tevo
2020-08-04 19:03:41,588 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [127]: NAME IS: tevo
2020-08-04 19:03:41,589 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [128]: ADDRESS IS: 192.168.1.24
2020-08-04 19:03:41,595 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [58]: Could not get metadata of container tevo because it was not found.

VlaoMao, 2020-08-04:
Huh.. Removed all Cura configs, recreated printer profile and all works

closed 2022-01-07 by pforrmi