VMaxx/RepetierIntegration #48: feature request: upload (optional with GROUP-NAME) and print

fajabird, 2020-11-23:
Currently I have to choose between print and save to repetier-server in the plugin options. If I choose "print" than I sometimes miss the option to quickly start the pring again, if I choose "save" then I need another step, i.e. load the repetier-server webUI - start print.
 Would it be possible to have both, i.e. save the file to the server and then let it print from cura, so in case the print did not go well I can restart it from the server rather than starting over (e.g. open cura, load STL, slice, print)? It would be great if GROUP-NAMES would be supported by the plugin, so upon uploading gcode to the printer the user is prompted to select the GROUP. The groups can be queried easily like this:

Query groups from server:
http://localhost:3344/printer/api/<slug>?a=listModelGroups&apikey=<API key>>

Upload using group-name:
http://localhost:3344/printer/model/<slug>?a=upload&sess=<session key>&name=<file name>&group=<groupname>

VMaxx, 2021-03-11:
I'll see if I can get this to work.

VMaxx, 2021-03-12:
listModelGroups doesn't seem to be functioning right.