VMaxx/RepetierIntegration #49: Repetier Server Connection to Cura 4.8

westice2000, 2021-04-17:
Hello

I m trying to connect my Repetier server to Cura 4.8. I have taken a look at the repository and notice that the latest version there is 4.7... I have used the master branch and copied to the plugin folder

Following a restart of Cura, if I select the printer I wish to connect I see the "Connect Repetier" button. I complete the required input:

Rpetier Server Name
Correct IP address
Add my Repetier server ID and password

When selecting the "Get Printers" button, I can see the printer I wish to attach, and press OK

From here I select the created server link and enter the API key from my server, and this gives me the following error:

"Repetier printer name hasn't been selected, please edit and click get printers and choose the correct name from the drop down list"

I cannot go any further. Selecting the "Open in browser" button does take me to the printer though my web browser, but the "Connect" option is greyed out

Thanks for your support

bradjshannon, 2021-04-21:
Same issue here. Looks like most of the issues reported in github are on this subject. Some folks have got it working by being VERY CAREFUL about how the printer is named when you set it up, in both places. 

Or something like that? See the discussion here: https://github.com/VMaxx/RepetierIntegration/issues/33f

I'm about to give this a try.

westice2000, 2021-04-22:
I cannot find the "machine_instances" file in my Cura (Mac), Which I should delete. It will now be displayed to me. I cannot connect. I have now made an update to 4.9
[image]
[image]
[image]

VMaxx, 2021-04-23:
it should get the name from repetier.  Make sure it is defined there first.

paulcafuk, 2021-05-11:
Like Westice2000 I'm on a Mac and can't find the machine_instances file anywhere on my Mac. Any clue where it's at?

paulcafuk, 2021-05-12:
Gotcha! I’ll keep trying to get it to work!

DavidWildJnr, 2021-05-22:
For those looking for the machine_instance folder it was here for me ->

/Users/<myUserFolder>/Library/Application Support/cura/4.9
Replace <myUserFolder> with your own.

However , still no look getting the plugin to work.

westice2000, 2021-05-22:
Thank you for the information. I tried it again but I couldn't get it to work either.

bourneash, 2021-07-27:
@westice2000 @paulcafuk - Downgrading to Cura 4.8.0 fixed this issue for me, I was trying 4.10 and had the same issue.  https://github.com/Ultimaker/Cura/releases/tag/4.8.0

Geraldf, 2021-08-12:
> @westice2000 @paulcafuk - Downgrading to Cura 4.8.0 fixed this issue for me, I was trying 4.10 and had the same issue. https://github.com/Ultimaker/Cura/releases/tag/4.8.0

I have the the same issue with Cura 4.10 on a Mac as well, I can see the printer, but I cannot select it. 
Looking at the log file, there is also an error reported in the DiscoverRepetierAction.qml:722 file, which might be related. 
This is the location, where the Compbox get filled wiht the result from the ReptierServer
`RepetierIntegration/DiscoverRepetierAction.qml:722: TypeError: Cannot read property 'key' of undefined`

[image]

VMaxx, 2021-08-12:
Geraldf do you have your ipaddress/server name correct there?  repe**o**tierserver.local appears to have an o in it?

VMaxx, 2021-08-13:
I managed to get access to replicate this.  I'll see if I can debug and figure out why the selection is dimmed out for macs and works fine for windows and linux.

VMaxx, 2021-08-13:
ok update on this is there is a bug in qtkit that impacts macs.  https://bugreports.qt.io/browse/QTBUG-89607 I'm looking into what needs to be done to fix it.

Geraldf, 2021-08-13:
> Geraldf do you have your ipaddress/server name correct there? repe**o**tierserver.local appears to have an o in it?

@VMaxx , you are right with the wrong servername, but the problem exists also with the correct servername as well as with the IP Address. :-)

VMaxx, 2021-08-13:
If I can't find a qtkit fix, I'll change it into a listbox.

VMaxx, 2021-08-14:
Ok macOS fix is in, I got it working under the latest Big Sur so it should be good to go.  Its not the nice combo box looking thing but it is functioning when you click the listbox looking widget.

Geraldf, 2021-08-14:
> Ok macOS fix is in, I got it working under the latest Big Sur so it should be good to go. Its not the nice combo box looking thing but it is functioning when you click the listbox looking widget.

HI @VMaxx , thanks for fixing the first part. Now I can select the printer, and move to the next screen. But there is a new issue:
See the screenshot, and I do not know how to continue. 
[image]

VMaxx, 2021-08-14:
The repetieriD is blank there, you need to pick the ID from the list box area.  If you pick a blank name it will continue, but its obviously not the real name in the repetier server.

Geraldf, 2021-08-15:
> The repetieriD is blank there, you need to pick the ID from the list box area. If you pick a blank name it will continue, but its obviously not the real name in the repetier server. So click edit, hit the dropdown again, then pick the real name. It is also good to have the name to match your repetier printer name.

HI @VMaxx, understood, but clicking on the "Any" in the list box, does not fill the reptierID field.

VMaxx, 2021-08-16:
you can open the cura.cfg file under Users/<username>/Library/Application Support/cura/4.10
You should see a [Repetier] section somewhere and underneath is manual_instances
for example yours would be something like this:
```
[Repetier]
manual_instances = {"Any": {"address": "192.168.1.250", "port" : 3344, "path":  "/", "useHttps": false, "userName": "vmaxx", "password": "password", "repetier_id": "Any"}}
```
The missing piece is "repetier_id": "Any" for you.  Adjust it then close cura and restart it to see if it fixes things.

Siox0911, 2022-01-12:
Yes, in Logs I found this...
[image]

Siox0911, 2022-01-12:
I have changed your code localy from
`
        url = QUrl("http://" + base_url + "/printer/info")
`
to
`        if("http" in base_url):
            url = QUrl(base_url + "/printer/info")
        else:
            url = QUrl("http://" + base_url + "/printer/info")
`
This will work. But your plugin will not work on Cura 4.12. And Python grrrrr not my fav language.

VMaxx, 2022-01-13:
in your screen shot you have it http// the colon is missing.  Although it
is not expecting that it just wants your server name no http or https so it
should just be your repetierserver2 there.  once you remove that and only
put the name it should work.  There is a catch if your dns doesn't
recognize the name then you'll have to put the IP address there.

Siox0911, 2022-01-14:
You read this log wrong. Read the 2nd line in the screenshot. This first httpRequest is working, but the further requests not. Thats the reason why I have changed the code. But this i cant connect! 
[image]

VMaxx, 2022-01-14:
I was referring to this line:
2022-01-12 17:22:01,492 - DEBUG - [MainThread]
RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]:
getPrinterList:http://http//repetierserver2:3344//printer/info
there are two issues here: the "http//" in front of repetierserver2 and the
"//" after the 3344 port.
http://http//repetierserver2:3344//printer/info
What is curious is that the previous getPrinterList at line 2 is correct.
the base_url isn't supposed to have http in it so your code shouldn't
really run.
Check your %appdata%/cura/4.8/cura.cfg file and make sure the manual
instances entry is correct, sometimes cura doesn't edit it properly.
it should look something like this, except username and password.
manual_instances = {"Ronnys_3DDrucker": {"userName": "Ronny", "password":
"Ronny's password", "useHttps": false, "address": "repetierserver2",
"path": "/", "port": 3344, "repetier_id": "Ronnys_3DDrucker"}}

Siox0911, 2022-01-14:
`manual_instances = {"Ronnys_3DDrucker": {"address": "repetierserver2", "port": 3344, "path": "/", "useHttps": false, "userName": "", "password": "", "repetier_id": "Ronnys_3DDrucker1"}}`

All is fine.

Siox0911, 2022-01-14:
Additional... With my code change, that are the new log entries. What I have done. Start Cura, go to the printer settings and start one time the configuration for RepetierIntegration. 

```
2022-01-14 17:28:26,311 - DEBUG - [JobQueueWorker [1]] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
2022-01-14 17:28:27,007 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [213]: Error check finished, result = False, time = 5.7s
2022-01-14 17:28:32,854 - INFO - [JobQueueWorker [0]] SettingsGuide2.CuraSettingsGuide.load_all [174]: Finished loading Settings Guide articles.
2022-01-14 17:28:36,437 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.03194308280944824 seconds
2022-01-14 17:29:10,114 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///C:/Users/lz_sp/AppData/Roaming/cura/4.12/plugins/RepetierIntegration/RepetierIntegration/DiscoverRepetierAction.qml:296:21: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
2022-01-14 17:29:10,122 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [62]: Could not get metadata of container Ronnys_3DDrucker because it was not found.
2022-01-14 17:29:10,124 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [62]: Could not get metadata of container Ronnys_3DDrucker because it was not found.
2022-01-14 17:29:10,126 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [62]: Could not get metadata of container Ronnys_3DDrucker because it was not found.
2022-01-14 17:29:10,128 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [62]: Could not get metadata of container Ronnys_3DDrucker because it was not found.
2022-01-14 17:29:10,128 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [260]: getPrinterList:http://repetierserver2:3344//printer/info
2022-01-14 17:29:10,131 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///C:/Users/lz_sp/AppData/Roaming/cura/4.12/plugins/RepetierIntegration/RepetierIntegration/DiscoverRepetierAction.qml:158: ReferenceError: objectListFooter is not defined
2022-01-14 17:29:10,133 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///C:/Users/lz_sp/AppData/Roaming/cura/4.12/plugins/RepetierIntegration/RepetierIntegration/DiscoverRepetierAction.qml:709: ReferenceError: propertyProvider is not defined
2022-01-14 17:29:10,172 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [473]: http://repetierserver2:3344//printer/info
2022-01-14 17:29:10,172 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [478]: http://repetierserver2:3344//printer/info
2022-01-14 17:29:10,172 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [479]: {'apikey': 'Ronny_3DPrinterApiKey', 'name': 'Repetier-Server Pro', 'printers': [{'active': True, 'name': 'TronXY X1', 'online': 0, 'slug': 'TronXY_X2'}, {'active': True, 'name': 'Ronnys 3D-Drucker', 'online': 0, 'slug': 'Ronnys_3DDrucker1'}], 'servername': 'Devi-50', 'serveruuid': 'ce57ee80-a693-4613-8a48-5a27df3ef862', 'version': '1.2.0'}
2022-01-14 17:29:10,173 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: DiscoverRepetierAction: printers: 2
2022-01-14 17:29:10,173 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [489]: Slug: TronXY_X2
2022-01-14 17:29:10,173 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [489]: Slug: Ronnys_3DDrucker1
2022-01-14 17:29:10,173 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [493]: DiscoverRepetierAction: apikey: Ronny_3DPrinterApiKey
2022-01-14 17:29:11,528 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [62]: Could not get metadata of container Ronnys_3DDrucker because it was not found.
2022-01-14 17:29:11,530 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [62]: Could not get metadata of container Ronnys_3DDrucker because it was not found.
2022-01-14 17:29:11,530 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [260]: getPrinterList:http://repetierserver2:3344//printer/info
2022-01-14 17:29:11,539 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [473]: http://repetierserver2:3344//printer/info
2022-01-14 17:29:11,539 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [478]: http://repetierserver2:3344//printer/info
2022-01-14 17:29:11,539 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [479]: {'apikey': 'Ronny_3DPrinterApiKey', 'name': 'Repetier-Server Pro', 'printers': [{'active': True, 'name': 'TronXY X1', 'online': 0, 'slug': 'TronXY_X2'}, {'active': True, 'name': 'Ronnys 3D-Drucker', 'online': 0, 'slug': 'Ronnys_3DDrucker1'}], 'servername': 'Devi-50', 'serveruuid': 'ce57ee80-a693-4613-8a48-5a27df3ef862', 'version': '1.2.0'}
2022-01-14 17:29:11,540 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: DiscoverRepetierAction: printers: 2
[... cut ...]
2022-01-14 17:29:33,294 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: DiscoverRepetierAction: printers: 2
2022-01-14 17:29:33,294 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [489]: Slug: TronXY_X2
2022-01-14 17:29:33,294 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [489]: Slug: Ronnys_3DDrucker1
2022-01-14 17:29:33,295 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [493]: DiscoverRepetierAction: apikey: Ronny_3DPrinterApiKey
```

Siox0911, 2022-01-14:
This is the url in repetierserver
`http://repetierserver2:3344/#!/printer/Ronnys_3DDrucker1/console`
So ID is correct...
Port is correct
The name I think, this is a name... 

I have changed the printer name to the printer id. Without any effects.

VMaxx, 2022-01-14:
in the log this: "http://repetierserver2:3344//printer/info"
has an extra slash
try changing your code to:
if("http" in base_url): url = QUrl(base_url + "/printer/info") else: url = QUrl("http://" + base_url + "printer/info")

VMaxx, 2022-01-17:
yeah its just cura does odd things with the names etc.  I'd delete it in
cura restart and match it to the repetier name.