VMaxx/RepetierIntegration #50: Cura 4.1 Integration on Mac Pro(2012)

kn4ycd, 2021-09-05:
I've installed the 4.1 branch plugin in the <home>/Library/Application Support/cura/4.1/plugins folder, as "RepetierIntegration".

I cannot get Cura to recognize it in the printer dialog at all.  I've read the docs several times, and I must be missing something here.

When I start Cura, it is creating the _pycache_ folder on first run, but that is all I'm seeing as far as Cura using the plugin.

VMaxx, 2021-09-05:
Once installed you will see an extra button to configure under manage printers.  It will say Connect Repetier.  Click add and match the name you put in repetier and fill out the dialog.Sent from my Verizon, Samsung Galaxy smartphone

kn4ycd, 2021-09-05:
Right - not seeing it at all.  Don’t get that button in any way.

VMaxx, 2021-09-05:
You'll have to look at the log file and check for errors then.  Post it if you can.

VMaxx, 2021-09-05:
Plugin should be in your Users\<your username>\Library\Application Support\cura\4.10\plugins folder.

VMaxx, 2021-09-05:
log will be in the 4.10 folder.

kn4ycd, 2021-09-05:
Correct, it is.

VMaxx, 2021-09-10:
can you post the contents of said log file?

kn4ycd, 2021-09-10:
Attached is the log file.