VMaxx/RepetierIntegration #52: Error on Cura 4.11 when accessing the plugin-settings

fajabird, 2021-09-29:
system: Manjaro Linux (up-to-date)

Traceback:

Traceback (most recent call last):
  File "/home/artur/.local/share/cura/4.11/plugins/RepetierIntegration/DiscoverRepetierAction.py", line 532, in _onRequestFinished
    stream_url = json_data["webcams"][0]["dynamicUrl"].replace("127.0.0.1",re.findall( r'[0-9]+(?:\.[0-9]+){3}', reply.url().toString())[0])
IndexError: list index out of range

Logs:

Thread 0x00007f133bfff640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 2547 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1735 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f135083f640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.9/threading.py", line 910 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f1351040640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1485 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f1351841640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1359 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f1352042640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 910 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f1352843640 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 910 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f13757fa640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/usr/lib/python3.9/threading.py", line 910 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f1375ffb640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/usr/lib/python3.9/threading.py", line 910 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f138e7fc640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f138effd640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f138f7fe640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Thread 0x00007f138ffff640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.9/threading.py", line 973 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 930 in _bootstrap

Current thread 0x00007f13bb107740 (most recent call first):
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 396 in _logInfoWidget
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/usr/bin/cura", line 159 in exceptHook
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 393 in exec_
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 867 in run
  File "/usr/bin/cura", line 233 in <module>

VMaxx, 2021-09-30:
What version of repetier are you using?

fajabird, 2021-09-30:
its 1.0.3 free (apollo-84)

fajabird, 2021-09-30:
I will do the update to latest version 1.1.2 and report again. thanks for the hint.

fajabird, 2021-09-30:
also: it may not be specific to Cura 4.11 only, but I know it was not happening on 4.8 when I initially set up the plugin.