VMaxx/RepetierIntegration #53: Printer not selectable

TAOGde, 2021-10-17:
After replacing my mainboard on the Artillery Sidewinder X1 it does not work anymore.
U can see that the Printerlist is not clickable and useless at this point. It works with my old board without any problem.

I deleted all settings on the MAC-OS System ( 11.6 ) and started from zero but no luck.

[image]

Any suggestion?

Regards Torsten

TAOGde, 2021-10-18:
> when you go to your repeitier web page does it let you send commands to the printer there?

On my Raspberry Touchscreen and Browser all works fine.

VMaxx, 2021-10-21:
I updated my mac to 11.6 to test and downloaded cura 4.11 to verify.  I think you need to downloaded the 4.10 update with the mac fix, from the screen shot you're using a pre-bugfix version that uses the older QT dropdown versus the new QT2 listbox to select the printer.