VMaxx/RepetierIntegration #55: back and forth between input API and printer name not selected

clynch213, 2022-01-02:
I'm using cura 4.12 on windows and cura asks me to input the API after putting everything in. Then it give me the error that repetier printer name not selected. So I click edit, select the printer name, then click OK. After clicking ok, the API disappears and I'm back to the beginning. Any help would be greatly appreciated.
[image]

clynch213, 2022-01-03:
Please Ignore me. I got it working once I set up a new printer profile instead of connecting one of my existing profiles.

Siox0911, 2022-01-12:
Nothing to ignore. It is not working in Cura. 4.12

Neokil, 2022-01-30:
I have the same issue. I tried it with the master-branch and also the 4.7-branch and always the same issue, I add a new repetier-server, when I click on "get Printers" the list gets populated, I select one and click ok and after filling in the API-Key it tells me to go back and select a printer. (the message of the original post)

clynch213, 2022-01-30:
The way I fixed mine with the master branch was to restart cura and then add a new printer to cura then connect to repetier. Hope this helps. I am using cura 4.12

blue0ceanby, 2022-02-03:
Hello VMaxx,
thank you for your good work Unfortunately, the error comes back in version 4.13.

JimB1707, 2022-02-16:
Hello I am having the same issue as @blue0ceanby 

I have removed the printer from Cura & restarted (& Repetier server for good measure) as per @clynch213 resolution, but I am still unable to connect in 4.13

Thanks

Jim

VMaxx, 2022-02-19:
Under %appdata%\cura\4.13 there should be a cura.cfg and cura.log look through the log for errors that you can post.  In the cura.cfg copy the [Repetier] section out and post it, remove any personal things like passwords and apikeys and replace them with ### and post it so I can maybe come up with a fix.

JimB1707, 2022-02-22:
Hi There

 

Thanks for reaching back

 

Please find attached the cura CFG & LOG files

 

Kind regards

 

Jim Bruce

alez0101, 2022-03-15:
Hello. I'm also experiencing this issue.

My cura.cfg:
data	sort
2022-03-15 14:39:35,097 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:39:38,372 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:39:40,449 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:40,449 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:40,449 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:47,085 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:39:47,089 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [469]: http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:39:47,089 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [474]: http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:39:47,089 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [475]: {'apikey': '0d7a0ed3-474c-4033-bacd-7addc0df1bf3', 'name': 'Repetier-Server Free', 'printers': [{'active': True, 'name': 'AnyCubic 4Max Pro 2.0', 'online': 1, 'slug': 'AnyCubic_4Max_Pro_20'}, {'active': True, 'name': 'Ender 3', 'online': 1, 'slug': 'Ender_3'}], 'servername': 'Tartarus-89', 'serveruuid': 'c5391bbe-3a99-4ec6-b053-2d2d422bbcf9', 'version': '1.3.0'}	DEBUG - [M
2022-03-15 14:39:47,090 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [481]: DiscoverRepetierAction: printers: 2	DEBUG - [M
2022-03-15 14:39:47,090 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: Slug: AnyCubic_4Max_Pro_20	DEBUG - [M
2022-03-15 14:39:47,090 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: Slug: Ender_3	DEBUG - [M
2022-03-15 14:39:47,091 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [489]: DiscoverRepetierAction: apikey: 0d7a0ed3-474c-4033-bacd-7addc0df1bf3	DEBUG - [M
2022-03-15 14:39:50,895 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:39:50,899 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [100]: Repetier_ID: Ender_3	DEBUG - [M
2022-03-15 14:39:50,900 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [127]: NAME IS: Ender 3	DEBUG - [M
2022-03-15 14:39:50,900 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [128]: ADDRESS IS: 192.168.86.35	DEBUG - [M
2022-03-15 14:39:52,601 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:39:53,222 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:53,222 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:53,659 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:40:00,506 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:40:00,925 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.012433052062988281 seconds	DEBUG - [M
2022-03-15 14:40:53,973 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:40:53,978 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [469]: http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:40:53,978 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [474]: http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:40:53,979 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [475]: {'apikey': '0d7a0ed3-474c-4033-bacd-7addc0df1bf3', 'name': 'Repetier-Server Free', 'printers': [{'active': True, 'name': 'AnyCubic 4Max Pro 2.0', 'online': 1, 'slug': 'AnyCubic_4Max_Pro_20'}, {'active': True, 'name': 'Ender 3', 'online': 1, 'slug': 'Ender_3'}], 'servername': 'Tartarus-89', 'serveruuid': 'c5391bbe-3a99-4ec6-b053-2d2d422bbcf9', 'version': '1.3.0'}	DEBUG - [M
2022-03-15 14:40:53,979 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [481]: DiscoverRepetierAction: printers: 2	DEBUG - [M
2022-03-15 14:40:53,979 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: Slug: AnyCubic_4Max_Pro_20	DEBUG - [M
[... cut ...]
2022-03-15 14:42:41,304 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:42:41,304 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:42:44,507 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.013382434844970703 seconds	DEBUG - [M
2022-03-15 14:43:13,284 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M

Ikswoknib, 2022-04-13:
Any efforts on this issue ? Having the same Problems....

Siox0911, 2022-04-13:
Not for me. So I can't use it. Current my extruder is completely disassembled, so I have no pressure. 😂

alez0101, 2022-04-13:
I haven't seen any changes. I still save my gcode to a shared directory on my Repetier.

VMaxx, 2022-04-14:
yeah things have been difficult at home as of late.

alez0101, 2022-04-14:
> yeah things have been difficult at home as of late.
> […](#)

I'm sorry to hear that. Hopefully our comments weren't  taken as rude. Wishing you all the best.

Siox0911, 2022-04-14:
> Under %appdata%\cura\4.13 there should be a cura.cfg and cura.log look through the log for errors that you can post. In the cura.cfg copy the [Repetier] section out and post it, remove any personal [...]

@alez0101 Do you understand this. And please replay you post here: https://github.com/VMaxx/RepetierIntegration/issues/55#issuecomment-1068507112

alez0101, 2022-04-14:
Hello. I'm also experiencing this issue.

My cura.cfg:
data	sort
2022-03-15 14:39:35,097 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:39:38,372 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:39:40,449 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:40,449 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:40,449 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:47,085 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:39:47,089 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [469]: http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:39:47,089 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [474]: http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:39:47,089 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [475]: {'apikey': '##########', 'name': 'Repetier-Server Free', 'printers': [{'active': True, 'name': 'AnyCubic 4Max Pro 2.0', 'online': 1, 'slug': 'AnyCubic_4Max_Pro_20'}, {'active': True, 'name': 'Ender 3', 'online': 1, 'slug': 'Ender_3'}], 'servername': 'Tartarus-89', 'serveruuid': 'c5391bbe-3a99-4ec6-b053-2d2d422bbcf9', 'version': '1.3.0'}	DEBUG - [M
2022-03-15 14:39:47,090 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [481]: DiscoverRepetierAction: printers: 2	DEBUG - [M
2022-03-15 14:39:47,090 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: Slug: AnyCubic_4Max_Pro_20	DEBUG - [M
2022-03-15 14:39:47,090 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: Slug: Ender_3	DEBUG - [M
2022-03-15 14:39:47,091 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [489]: DiscoverRepetierAction: apikey: ##########	DEBUG - [M
2022-03-15 14:39:50,895 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:39:50,899 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [100]: Repetier_ID: Ender_3	DEBUG - [M
2022-03-15 14:39:50,900 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [127]: NAME IS: Ender 3	DEBUG - [M
2022-03-15 14:39:50,900 - DEBUG - [MainThread] RepetierIntegration.RepetierOutputDevice.__init__ [128]: ADDRESS IS: 192.168.86.35	DEBUG - [M
2022-03-15 14:39:52,601 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:39:53,222 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:53,222 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:39:53,659 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://http//192.168.86.35:3344//printer/info	DEBUG - [M
2022-03-15 14:40:00,506 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:40:00,925 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.012433052062988281 seconds	DEBUG - [M
2022-03-15 14:40:53,973 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [256]: getPrinterList:http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:40:53,978 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [469]: http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:40:53,978 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [474]: http://192.168.86.35:3344/printer/info	DEBUG - [M
2022-03-15 14:40:53,979 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [475]: {'apikey': '##########', 'name': 'Repetier-Server Free', 'printers': [{'active': True, 'name': 'AnyCubic 4Max Pro 2.0', 'online': 1, 'slug': 'AnyCubic_4Max_Pro_20'}, {'active': True, 'name': 'Ender 3', 'online': 1, 'slug': 'Ender_3'}], 'servername': 'Tartarus-89', 'serveruuid': 'c5391bbe-3a99-4ec6-b053-2d2d422bbcf9', 'version': '1.3.0'}	DEBUG - [M
2022-03-15 14:40:53,979 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [481]: DiscoverRepetierAction: printers: 2	DEBUG - [M
2022-03-15 14:40:53,979 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [485]: Slug: AnyCubic_4Max_Pro_20	DEBUG - [M
[... cut ...]
2022-03-15 14:42:41,304 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:42:41,304 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M
2022-03-15 14:42:44,507 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.013382434844970703 seconds	DEBUG - [M
2022-03-15 14:43:13,284 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [465]: Received no or empty reply	DEBUG - [M

alez0101, 2022-04-14:
I believe I gathered the correct logs, maybe not. I updated to remove the code from the comment.

VMaxx, 2022-04-20:
> 

On both of your printers the name and the slug aren't the same.  I'd try changing the name to match the slug:
Change 'AnyCubic 4Max Pro 2.0' to 'AnyCubic_4Max_Pro_20'
and 
'Ender 3' to 'Ender_3'
On your ip address or host name for the printers only use the IP not the http form:
For example:
It appears you have http//192.168.1.35:3344/
It should look something like this:
[image]

VMaxx, 2022-04-21:
I was working on this some more and found the specific issue.  I'll be posting a fix soon.

Ikswoknib, 2022-08-06:
Hi,
I am still having this issue. I tried Cura 5.0.0 with the RepetierIntegration 5.0 and also with the master branch. Both have the same problem with that back and fourth ting.

I also tried the Cura 5.1 with both of the RepetierIntegration Versions but still the same....

Any Ideas ?

Ikswoknib, 2022-08-06:
I got it to work now.
I edited the configuration file of my printer in "Users\XXX\AppData\Roaming\cura\5.1\machine_instances"
in the file i deleted the following lines
repetier_auto_print = false
post_processing_scripts = 
repetier_show_camera = true

after that it works.
best regards