VMaxx/RepetierIntegration #56: compatibility to Cura 5.0 - no installation possible

Kasu81, 2022-04-28:
Hello,

Cura 5 beta es released.
I have tried several times to install the plugin , with the already known procedures. But it didn't work.
The functions are not appearing in the printer manager menu.

In regard of this Topic in the Cura Forum -->
It seems, that the plugin must be updated to get access to the new version of Cura.(Interface/Code--> I am not a subject matter expert :-)  )

https://community.ultimaker.com/topic/40288-ultimaker-cura-50-beta/

Are there actions ongoing to get the plugin compatible

br
Kasu

Chokulat, 2022-05-21:
I really love the new Cura 5.0.  But I REALLY  love being able to hit "print with repeater".   I really hope someone can update that file to make it possible.

VMaxx, 2022-05-21:
Thanks I'll see what I can do to get it going on 5.Sent from my Verizon, Samsung Galaxy smartphone

Darksk0pe, 2022-05-25:
+1 :) please make it possible

VMaxx, 2022-05-29:
ok a version is out there to give a try.

Chokulat, 2022-05-29:
ok.  Shane, you are AWESOME.  I was able to connect to my Repetier server, and almost cried.  Thank you for making an awesome plugin, and for updating it.

Darksk0pe, 2022-06-04:
> 

Same Problem .. Artillery genius PRO   Printer is displayed but not selectable

VMaxx, 2022-06-05:
Check your config file under machine_instances make sure the data in there is correct.  Also check your cura.log for error messages that might help point to what may be wrong.

jameslkey, 2022-06-06:
If it's the same issue I'm having it appears to be related to the PyQt implementation and not the plugin logic. I've tried mouse and keyboard shortcuts. I can not select the printer.

ThomasSlowhand, 2022-06-06:
I also have the same problem with the installtion under Cura 5.0. Printer list appears when clicking "Get printers" but none is selectable. Instance name has been chosen being the same like the printers in Cura 5.0. Please see attached screenshot.
To take care for the waring regarding the bug in Cura both printers have been installd from scratch and not renames. Can it be an issue that both printer have blanks in their names ("Ultimaker 2+"?
[image]

VMaxx, 2022-06-06:
I'll see what I can do to test.  I apparently accidentally copied the octoprint qml up that isn't used so I need to delete it.

VMaxx, 2022-06-06:
Ok I posted an update to fix the selection combobox.

soundylinz, 2022-06-28:
Thank you for the update! Now the plugin seems to work fine with Cura 5.0.0 as well. However, I did not set up any new printers, but worked with the existing printers. Hopefully there are no unexpected problems.