VMaxx/RepetierIntegration #57: Cura crashes when selecting a connection

G4Cab, 2022-06-11:
When i try to connect to my repetier, i'll get always this error.

Using Cura 5.0 release and Repetierintegration from today ( 64a68f0 )

Errormessage:

Traceback (most recent call last):
  File "C:\Users\G4Cab\AppData\Roaming\cura\5.0\plugins\RepetierIntegration\DiscoverRepetierAction.py", line 573, in _onRequestFinished
    stream_url = json_data["webcams"][0]["dynamicUrl"].replace("127.0.0.1",re.findall( r'[0-9]+(?:\.[0-9]+){3}', reply.url().toString())[0])
IndexError: list index out of range

[image]

[image]

VMaxx, 2022-06-14:
look for this line in your cura.log and post it.
DiscoverRepetierAction: webcams:

VMaxx, 2022-06-17:
what do you have for dynamicurl.
You can find it in the repetier setup here
[image]
:

G4Cab, 2022-06-18:
Dynamic http://192.168.101.1:8090/deadpool.mjpeg

Static http://192.168.101.1:8090/deadpool.jpg

Camerastream is provided from an ffmpegserver

VMaxx, 2022-06-18:
OK I updated the DiscoverRepetierAction.py file.  Please redownload it and try it again then send the log again.
I suspect there is something missing in this part "reply.url().toString())[0]" so I am having it log what is in that value before it tries to use it later.  I can then try to make a fix once I see what you're getting.

G4Cab, 2022-06-19:
```
2022-06-19 16:48:21,607 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [569]: DiscoverRepetierAction: webcams: 1
2022-06-19 16:48:21,608 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [572]: DiscoverRepetierAction: Checking streamurl
2022-06-19 16:48:21,609 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [573]: DiscoverRepetierAction: reply_url: PyQt6.QtCore.QUrl('http://sapphirepro-repetier:3344//printer/api/SapphirePro?a=getPrinterConfig&apikey=blablablablablablablablablablablablablablabla')
2022-06-19 16:48:21,623 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2022-06-19 16:48:21,623 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2022-06-19 16:48:21,624 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Users\G4Cab\AppData\Roaming\cura\5.0\plugins\RepetierIntegration\DiscoverRepetierAction.py", line 574, in _onRequestFinished
2022-06-19 16:48:21,624 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     stream_url = json_data["webcams"][0]["dynamicUrl"].replace("127.0.0.1",re.findall( r'[0-9]+(?:\.[0-9]+){3}', reply.url().toString())[0])
2022-06-19 16:48:21,625 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: IndexError: list index out of range
2022-06-19 16:48:22,066 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [262]: getPrinterList:base_url:http://Deadpool450-repetier:3344/
2022-06-19 16:48:22,067 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [264]: getPrinterList:http://deadpool450-repetier:3344/printer/info
2022-06-19 16:48:22,085 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [495]: http://deadpool450-repetier:3344/printer/info
2022-06-19 16:48:22,087 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [500]: http://deadpool450-repetier:3344/printer/info
2022-06-19 16:48:22,087 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [501]: {'apikey': 'blablablablablablablablablablablablablablabla', 'name': 'Repetier-Server Pro', 'printers': [{'active': False, 'name': 'EasyThreeD Nano', 'online': 0, 'slug': 'EasyThreeD_Nano'}, {'active': False, 'name': 'Deadpool450', 'online': 0, 'slug': 'Deadpool450'}], 'servername': 'Deadpool450', 'serveruuid': '9a3879ae-6a6f-4696-b42c-5231a21b1d2f', 'version': '1.3.0'}
2022-06-19 16:48:22,088 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [507]: DiscoverRepetierAction: printers: 2
2022-06-19 16:48:22,088 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: EasyThreeD_Nano
2022-06-19 16:48:22,090 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: Deadpool450
2022-06-19 16:48:22,090 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [515]: DiscoverRepetierAction: apikey: blablablablablablablablablablablablablablabla
2022-06-19 16:48:22,582 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [495]: http://sapphirepro-repetier:3344//printer/api/SapphirePro?a=listModelGroups&apikey=blablablablablablablablablablablablablablabla
2022-06-19 16:48:22,583 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [528]: http://sapphirepro-repetier:3344//printer/api/SapphirePro?a=listModelGroups&apikey=blablablablablablablablablablablablablablabla
2022-06-19 16:48:22,584 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [529]: {'groupNames': ['#'], 'ok': True}
2022-06-19 16:48:22,585 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [534]: DiscoverRepetierAction: groupNames: 1
2022-06-19 16:48:22,586 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [538]: groupName: #

```

VMaxx, 2022-06-23:
I uploaded another change to try.

G4Cab, 2022-06-25:
And here we crash again:

`2022-06-25 18:49:54,465 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [569]: DiscoverRepetierAction: webcams: 1
2022-06-25 18:49:54,466 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [572]: DiscoverRepetierAction: Checking streamurl
2022-06-25 18:49:54,466 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [573]: DiscoverRepetierAction: reply_url: PyQt6.QtCore.QUrl('http://sapphirepro-repetier:3344//printer/api/SapphirePro?a=getPrinterConfig&apikey=blablablablablablablablablablablablablablabla')
2022-06-25 18:49:54,503 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2022-06-25 18:49:54,504 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2022-06-25 18:49:54,504 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Users\G4Cab\AppData\Roaming\cura\5.0\plugins\RepetierIntegration\DiscoverRepetierAction.py", line 577, in _onRequestFinished
2022-06-25 18:49:54,505 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     Logger.log("d", "DiscoverRepetierAction: stream_url: %s",stream_url)
2022-06-25 18:49:54,506 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: UnboundLocalError: local variable 'stream_url' referenced before assignment
2022-06-25 18:49:54,914 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [262]: getPrinterList:base_url:http://Deadpool450-repetier:3344/
2022-06-25 18:49:54,915 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [264]: getPrinterList:http://deadpool450-repetier:3344/printer/info
2022-06-25 18:49:54,947 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [495]: http://deadpool450-repetier:3344/printer/info
2022-06-25 18:49:54,948 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [500]: http://deadpool450-repetier:3344/printer/info
2022-06-25 18:49:54,948 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [501]: {'apikey': 'blablablablablablablablablablablablablablabla', 'name': 'Repetier-Server Pro', 'printers': [{'active': False, 'name': 'EasyThreeD Nano', 'online': 0, 'slug': 'EasyThreeD_Nano'}, {'active': True, 'name': 'Deadpool450', 'online': 0, 'slug': 'Deadpool450'}], 'servername': 'Deadpool450', 'serveruuid': '9a3879ae-6a6f-4696-b42c-5231a21b1d2f', 'version': '1.3.0'}
2022-06-25 18:49:54,949 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [507]: DiscoverRepetierAction: printers: 2
2022-06-25 18:49:54,949 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: EasyThreeD_Nano
2022-06-25 18:49:54,950 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: Deadpool450
2022-06-25 18:49:54,950 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [515]: DiscoverRepetierAction: apikey: blablablablablablablablablablablablablablabla
2022-06-25 18:49:55,429 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [495]: http://sapphirepro-repetier:3344//printer/api/SapphirePro?a=listModelGroups&apikey=blablablablablablablablablablablablablablabla
2022-06-25 18:49:55,432 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [528]: http://sapphirepro-repetier:3344//printer/api/SapphirePro?a=listModelGroups&apikey=blablablablablablablablablablablablablablabla
2022-06-25 18:49:55,435 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [529]: {'groupNames': ['#'], 'ok': True}
2022-06-25 18:49:55,437 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [534]: DiscoverRepetierAction: groupNames: 1
2022-06-25 18:49:55,440 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [538]: groupName: #`

VMaxx, 2022-06-25:
Ok I put another update.

G4Cab, 2022-06-26:
OK, now it don't crashes cura anymore. But i can't connect, same trouble like anytimes, get printers -> select printer -> OK -> insert Api-Key -> Repetier printer name hasn't been selected, please edit .....

If you want, i can give you teamviewer access

Here's the log:

```
2022-06-26 20:47:56,905 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [262]: getPrinterList:base_url:http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/
2022-06-26 20:47:56,906 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [264]: getPrinterList:http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:47:57,013 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [495]: http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:47:57,015 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [500]: http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:47:57,016 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [501]: {'apikey': 'blablablablablablablablablablablablablablabla', 'name': 'Repetier-Server Pro', 'printers': [{'active': False, 'name': 'EasyThreeD Nano', 'online': 0, 'slug': 'EasyThreeD_Nano'}, {'active': True, 'name': 'Deadpool450', 'online': 0, 'slug': 'Deadpool450'}], 'servername': 'Deadpool450', 'serveruuid': 'blablablablablablablablablablablablablablabla', 'version': '1.3.0'}
2022-06-26 20:47:57,017 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [507]: DiscoverRepetierAction: printers: 2
2022-06-26 20:47:57,017 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: EasyThreeD_Nano
2022-06-26 20:47:57,018 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: Deadpool450
2022-06-26 20:47:57,019 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [515]: DiscoverRepetierAction: apikey: blablablablablablablablablablablablablablabla
2022-06-26 20:47:59,205 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [262]: getPrinterList:base_url:http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/
2022-06-26 20:47:59,208 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [264]: getPrinterList:http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:47:59,230 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [495]: http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:47:59,234 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [500]: http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:47:59,237 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [501]: {'apikey': 'blablablablablablablablablablablablablablabla', 'name': 'Repetier-Server Pro', 'printers': [{'active': False, 'name': 'EasyThreeD Nano', 'online': 0, 'slug': 'EasyThreeD_Nano'}, {'active': True, 'name': 'Deadpool450', 'online': 0, 'slug': 'Deadpool450'}], 'servername': 'Deadpool450', 'serveruuid': 'blablablablablablablablablablablablablablabla', 'version': '1.3.0'}
2022-06-26 20:47:59,240 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [507]: DiscoverRepetierAction: printers: 2
2022-06-26 20:47:59,244 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: EasyThreeD_Nano
2022-06-26 20:47:59,247 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: Deadpool450
2022-06-26 20:47:59,250 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [515]: DiscoverRepetierAction: apikey: blablablablablablablablablablablablablablabla
2022-06-26 20:48:03,395 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [62]: Could not get metadata of container Deadpool450 because it was not found.
2022-06-26 20:48:03,396 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [62]: Could not get metadata of container Deadpool450 because it was not found.
2022-06-26 20:48:05,533 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.04510903358459473 seconds
2022-06-26 20:48:51,395 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [262]: getPrinterList:base_url:http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/
2022-06-26 20:48:51,396 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction.getPrinterList [264]: getPrinterList:http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:48:51,409 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [495]: http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:48:51,410 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [500]: http://deadpool450-repetier.golf-4-cab.dynv6.net:3344/printer/info
2022-06-26 20:48:51,411 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [501]: {'apikey': 'blablablablablablablablablablablablablablabla', 'name': 'Repetier-Server Pro', 'prin [... cut ...]
2022-06-26 20:48:55,923 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [507]: DiscoverRepetierAction: printers: 2
2022-06-26 20:48:55,926 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: EasyThreeD_Nano
2022-06-26 20:48:55,930 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [511]: Slug: Deadpool450
2022-06-26 20:48:55,933 - DEBUG - [MainThread] RepetierIntegration.DiscoverRepetierAction._onRequestFinished [515]: DiscoverRepetierAction: apikey: blablablablablablablablablablablablablablabla
```

VMaxx, 2022-06-28:
what version of repetier ar eyou using?

G4Cab, 2022-06-28:
> what version of repetier ar eyou using?
Repetier-Server Pro 1.3.0