VMaxx/RepetierIntegration #58: Cura crashes when trying to turn on extruder or bed from monitor tab

TTVert, 2022-07-29:
Repetier server: 1.4.1
Cura 5.1
Win10

The monitor tab does display current and accurate information I just cannot pre-heat w/ these buttons

Error Traceback:
```
Traceback (most recent call last):
  File "cura\CuraApplication.py", line 1118, in event
    return super().event(event)
  File "UM\Qt\QtApplication.py", line 500, in event
    event._function_event.call()
  File "UM\Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "C:\Users\Redacted\AppData\Roaming\cura\5.1\plugins\RepetierIntegration\RepetierOutputDevice.py", line 522, in _sendQueuedGcode
    self._sendCommandToApi("send", "&data={\"cmd\":\"" + gcode + "\"}")
  File "C:\Users\Redacted\AppData\Roaming\cura\5.1\plugins\RepetierIntegration\RepetierOutputDevice.py", line 542, in _sendCommandToApi
    command_request.setHeader(QNetworkRequest.ContentTypeHeader, "application/json")
AttributeError: type object 'QNetworkRequest' has no attribute 'ContentTypeHeader'
```

Log:
```
Thread 0x000029a8 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 5.1.0\share\cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000057e4 (most recent call first):
  File "threading.py", line 324 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001950 (most recent call first):
  File "threading.py", line 324 in wait
  File "threading.py", line 600 in wait
  File "C:\Program Files\Ultimaker Cura 5.1.0\share\cura\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000039ac (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005f18 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 5.1.0\share\cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00006244 (most recent call first):
  File "UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003ad0 (most recent call first):
  File "UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000010c0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002780 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000065a4 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00006190 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002f58 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005d9c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003790 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000101c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00003658 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 165 in exceptHook
  File "UM\Qt\QtApplication.py", line 414 in exec
  File "cura\CuraApplication.py", line 888 in run
  File "cura_app.py", line 239 in <module>

```

TTVert, 2025-03-31:
This still exists, also crashes when trying to pause or any other controls from within cura.

Dave

VMaxx, 2025-03-31:
free or paid version?

TTVert, 2025-03-31:
I have Repetier Pro 1.4.18 paid.  When this originally occurred years ago I was on free though.

VMaxx, 2025-03-31:
ok windows?

VMaxx, 2025-04-01:
I put an update out that should fix this.  For whatever reason, they decided to change an underlying network class to declutter the properties.  However, it breaks any old code.

TTVert, 2025-04-01:
> I put an update out that should fix this. For whatever reason, they decided to change an underlying network class to declutter the properties. However, it breaks any old code.

Thank you, I will install and report back.  This has happened for me from the first day I used this extension many years ago but just sort of dealt with it.

Dave

TTVert, 2025-04-01:
That worked a treat thanks much!!

closed 2025-04-01 by TTVert