VMaxx/RepetierIntegration #60: Cura 5.01.0 Pre-Heat

lewist01, 2022-09-01:
**Pre-Heat option for nozzle and bed causes this error. Then crashes Cura.**

Traceback (most recent call last):
  File "cura\CuraApplication.py", line 1118, in event
    return super().event(event)
  File "UM\Qt\QtApplication.py", line 500, in event
    event._function_event.call()
  File "UM\Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "C:\Users\**\AppData\Roaming\cura\5.1\plugins\RepetierIntegration\RepetierOutputDevice.py", line 522, in _sendQueuedGcode
    self._sendCommandToApi("send", "&data={\"cmd\":\"" + gcode + "\"}")
  File "C:\Users\**\AppData\Roaming\cura\5.1\plugins\RepetierIntegration\RepetierOutputDevice.py", line 542, in _sendCommandToApi
    command_request.setHeader(QNetworkRequest.ContentTypeHeader, "application/json")
AttributeError: type object 'QNetworkRequest' has no attribute 'ContentTypeHeader'

SCandan69, 2022-09-19:
Ändern Sie in RepetierOutputDevice.py (542) von
command_request.setHeader(**QNetworkRequest**.ContentTypeHeader, "application/json")

in 

command_request.setHeader(**QNetworkRequestKnownHeaders**.ContentTypeHeader, "application/json")

dann geht dieser Bereich wieder.

VMaxx, 2022-09-22:
I put this change in so hopefully that'll fix it.

lewist01, 2022-10-12:
> I put this change in so hopefully that'll fix it.

Thank you for the fix.. works a treat :-)