VMaxx/RepetierIntegration #62: Cura 5.2.1

giangi74it, 2022-12-07:
I put all details and the api key but cannot add on cura...

VMaxx, 2022-12-15:
what OS and repetier version?

giangi74it, 2022-12-15:
Last versionInviato da iPhone 14Pro 512Gb"Hai saputo creare e vivere quella mela che non hai mai morsicatoSteve Sei stato un esempio di vita per tutti noi"

Kiwi0ne, 2023-01-24:
I second this, I select the printer and shows up, then go to the next page to paste the API key and tells me that I have not selected the printer in the page before.  I am on Windows 11, Repetier 1.4.6.  Any help is appreciated.

VMaxx, 2023-01-26:
Check and make sure the name you give it in the  repeiter connect is the same as the name in cura otherwise it has issues.

VMaxx, 2025-08-06:
if you have a login required on your repetier server, you need to click on the advanced security and type that information in.
To test you can try using http://3d.local:3344/ and make sure you can login, if not, then certainly the app can't either.