VMaxx/RepetierIntegration #63: error while prining: 'Message' object has no attribute 'replace'

ATOM1138, 2023-03-24:
[image]

when trying to print Cura says ('Message' object has no attribute 'replace' ) and nothing happens. What could be the possible sources of the problem? I have an older version of cura 4.7 with repetier pro and an Ultimaker s3 that prints totally fine through Cura directly.

Any information regarding what could help in theory or how the plugin works is totally welcome !

VMaxx, 2023-04-01:
Make sure you use the version. That matches the cura version.

EmEiBee, 2025-04-06:
Hello everyone,
I'd like to reopen this issue.
It suddenly appeared. I had printed a different part just before, without any problems. I'd also printed this part several times before, also without any problems.
Suddenly, the generated gCode can no longer be transferred to the Repetier server, regardless of what type of part I slice.

[export-20250406-172535.zip](https://github.com/user-attachments/files/19622123/export-20250406-172535.zip)

And yes. I've restarted everything several times. The connection to the Repetier server hasn't changed and is working.

Any idea?

Greetz
Micha

VMaxx, 2025-04-07:
Does the error show in RepetierOutputDevice.py or DiscoverRepetierAction.py?

EmEiBee, 2025-04-07:
The CURA version is 5.9.1, and the Repetier version is currently 1.4.18 PRO.
Repetier doesn't seem to notice when the error occurs.

I had similar errors earlier today with other, very simple components. After some fiddling around, I discovered by pure chance that it had something to do with "Use Adjustment Layers." I unchecked it for fun, and it worked without errors. I checked it again, and the error reappeared.
Interestingly, once I send it completely to Repetier without adjustment layers, it also works with that option, but only as long as CURA remains open. If I restart CURA, the error immediately reappears.
This behavior can be reproduced at will.

VMaxx, 2025-04-07:
I updated the plugin to see if it'll fix this for you.
RepetierOutputDevice.py specifically.
You can update your install, and see if it fixes.

EmEiBee, 2025-04-07:
Wow... that was a lightning-fast response! Thank you very much!
But what do you mean by update? I don't want to update to the new 5.10.x. It's too buggy for me.
Or can I download the Python file from GIT and manually replace it with the existing one?

VMaxx, 2025-04-07:
yes you can manually replace the file.

EmEiBee, 2025-04-07:
I renamed the old file to .old, copied the new file from GIT into it and start CURA.
Now the behavior is slightly different. It doesn't really matter whether the "Use Adjustment Layers" option is enabled or not. In both cases, the error message appears several times in a row.
If you leave the option enabled and then click "Print with Repetier" several times in a row, it will eventually be transferred.
But (and this is a big but!):
The files arriving at Repetier are all different sizes, ranging from 3.4 to 39.4 megabytes in this attempt, and take very different times.
Something is completely wrong here.

[image]

VMaxx, 2025-04-08:
Ok I updated DiscoverRepetierAction.py see if that helps.

EmEiBee, 2025-04-08:
Thank you again for your quick help. But I'm sorry to ruin your day...
Nothing has changed. The behavior is identical, and different files are produced on the RS when you click "Transfer to RS" 5 to 15 times to execute the transfer.
Attached are the results again. The red box again indicates the file that was transferred manually and is the only correct file.

[image]

And here you can find the Archive with the gCode what RS see (Folder ./printer/GEEE/jobs on the RS) similar to the screenshot. I have only included the ASCII Files .g and .gin
Maybe that helps...

[jobs.zip](https://github.com/user-attachments/files/19643924/jobs.zip)

VMaxx, 2025-04-08:
If you have any pyc files you may need to delete those so that cura
recompiles them.. they are often in the __pycache__ folder.

EmEiBee, 2025-04-08:
...good thing we talked about it ^^ That's obviously knowledge a mere mortal like me doesn't have... so that could well be the problem.

I'll do that later. Then I'll try again...

EmEiBee, 2025-04-08:
I found the cache, but there's no CURA subfolder there. I assume that when PY is used from within CURA, a separate folder is created for the application, as is common with other programs...
There are only subfolders of other applications (Kodi, Libre Office, FreeCAD, etc.) and the PY program cache folder itself.

VMaxx, 2025-04-08:
What OS are you using?