VMaxx/RepetierIntegration #65: Cure & Repetier

Golf3gti1, 2023-09-28:
Hello 
I have the latest version of Cura 5.4.0 and I followed your instructions and unfortunately I did not get a connection to the repeater. When I enter the APi number, the Connect button is grayed out. And I can not connect to the repeater. Would appreciate some help

VMaxx, 2023-09-29:
My 3d printer has been sick so I've not been keeping up with Cura lately.
I'll have to take a look at it and see.  There's likely a version issue as
I haven't updated it since version 5.2.1

Golf3gti1, 2023-09-29:
Since would be super if it would funktinieren

VMaxx, 2023-10-03:
OK I tested it as is with 5.4 and it worked fine.
The most common issue is not giving the instance name and printer name in repetier server the same.  If you have it mismatched you'll have to delete it in Cura, and re do it to match.

EmEiBee, 2024-01-08:
I'll hang on to it:
In CURA 5.5 the plugin cannot be made to cooperate.
Although I have completely re-created the printer as recommended and can select the printer from the drop-down with "get printer" even after connecting to Repetier, when I enter the API key it then complains that the printer was supposedly not selected.
[image]
[image]

VMaxx, 2024-01-08:
Your printer name doesn't match the instance name.  One had a space the other an underscoreSent from my Verizon, Samsung Galaxy smartphone

img-2024-01-08-11-45-52.png (view on web)
img-2024-01-08-11-47-00.png (view on web)

EmEiBee, 2024-01-08:
> Your printer name doesn't match the instance name.  One had a space the other an underscore
You are right but meanwhile I have try it some times more including fresh installation of CURA and also with the same name for the instance as the printer is named... No success.
I can mark them via "Get Printers" and if I come back it is till there, but at the moment if I enter the API the printer is gone, no matter if the names match or not...

VMaxx, 2024-01-09:
Try deleting the %appdata%\cura folder and reinstall.  make sure instance
and printer name you have in repetier are the same avoid spaces.

EmEiBee, 2024-01-09:
... I will try the next days and call back then; busy with other stuff right now (flood, frost, ...)

VMaxx, 2024-01-10:
Good luck, hang in there.

VMaxx, 2024-07-04:
I found that printers in repetier that use underscore seem to cause issues.  However, when I name a printer with a space it is converted to an underscore internally and works.  So if you named your printer in repetier using an underscore, I'd rename it there with a space instead.
For example I used "ziggy test" in repetier, but in the cura dropdown it reads as "ziggy_test" and works.
When I used "ziggy_test" in repetier it shows in cura the same but doesn't work.

VMaxx, 2025-06-02:
> have the same issue! Name of the printer is the same but the conntection
> button is still greyed out :-/

I suspect it is all tied to the underscores.

Gregor568, 2025-06-03:
I have completely deleted the printer from the repeater server and created a new one with the name “Kobra”. In Cura itself, I have entered “Kobra” in the settings for both the instance and the printer name, but the software still reports back that the repeater printer name is not selected and I am still unable to connect.