VMaxx/RepetierIntegration #66: 'NoneType' object has no attribute 'state' when trying to "Print with Repertier

Sectorchan, 2024-03-01:
Hi,

I've downloaded the Master yesterday, the connection seems to work, because I have a blue checkmark on the printer. When I click Print with Repertier I get the Error: 'NoneType' object has no attribute 'state'

Cura Version: 5.6.0 
RepertierServer 1.4.15

VMaxx, 2024-03-05:
Do you have the pro version or free and what OS are you using?

RueFrank, 2024-07-03:
Hola
new install the Connection --> https://github.com/VMaxx/RepetierIntegration

VMaxx, 2024-07-04:
I've not been able to replicate the issue.  However, I just found that printers in repetier that use underscore seem to cause issues.  Strangely, when I name a printer with a space it is converted to an underscore internally and works fine.  So if you named your printer in repetier using an underscore, I'd rename it there with a space instead.

VMaxx, 2024-07-13:
I've found this pops up when the repetier server isn't actually connected.  If its not on, or the ip or name given isn't correct.  I'm working on adjusting the code so that it will at least log a message instead of this none message.