VMaxx/RepetierIntegration #67: I can't monitor the printer in Cura.

turbospielt, 2024-10-08:
Hey, I have the problem that if I installed the plugin I can send print jobs via Cura, but I can't monitor the printer in Cura. It says that I have to connect the printer first.

VMaxx, 2024-12-11:
I ran it on 5.9 now and is seems to be ok.  Clear out your *.pyc files and retry it.

issacsim, 2024-12-20:
I am having same issue with 5.9.0, getting the "please connect printer" message.
This is a new new plugin installation, can't find any files with "pyc" extension in the plugin folder.

Another issue, not sure if related, In "Connect to Repetier" window, I can't select any of the folders in the dropdown list, they populate with the correct folder names, but none will gets selected when I click on them.

I was able to send a print job to Repetier, not sure which folder it sent it to. Had to select All files to find it.

VMaxx, 2024-12-20:
Look in your log files and send me some snippets or attach them for me to look at it.